MarlinFirmware/Marlin #19409: [BUG] Wrong z height when homing with ABL after changing z offset

Liger0, 2020-09-15:
### Bug Description

If you change the z offset of the probe and you home, the z height will be wrong

### My Configurations

13 september bugfix 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5228589/Marlin.zip)

### Steps to Reproduce

1. Home first time (may be skipped)
2. Change the z offset
3. Store settings (may be skipped)
4. Home

**Expected behavior:** The home should set position at 5.00mm Z

**Actual behavior:** The home goes to 5.10mm Z. 
Shutting off the printer is required.

#### Additional Information

thinkyhead, 2020-10-16:
Homing with a probe?

Liger0, 2020-10-16:
Yes.

Liger0, 2020-10-27:
Updated with today's bugfix, still same result.

sazzwi, 2020-11-12:
Not sure if I'm in the right spot or not since my Z home offset is 0, unchanged.  I have used m206 to adjust the X and Y but left the Z alone.     Here is why I suspect this could be related.   When I do  full power reset followed a G28 (home), the nozzle comes to a stop the display indicates that the Z is 0.00.     When I do a (G29) ABL followed by a G28 the display indicates the Z is 2.42. I have noticed  that the Z value after ABL varies which could be tolerance in the probe, heating of the bed etc.   This might be related so I figure I'd share my experience.  How are you supposed to get a consistent first layer with this behavior?

sjasonsmith, 2020-11-29:
I suspect this is caused by `RESTORE_LEVELING_AFTER_G28`. If leveling is enabled after `G28` the nozzle will move to Z5, then leveling will be applied which likely adjusts the Z height slightly.

Can you test this again, and run `M420`, does it report that leveling is enabled?

sjasonsmith, 2020-11-30:
Also, please provide _exact_ steps to reproduce.

I would like to see:
- Values from `M851` before first homing
- Exact G-code used to reproduce (including numbers)
- Output from `M420` and `M114` after reproducing the issue.

sjasonsmith, 2020-12-01:
Tagging @Liger0 to make sure they saw my requests for information above.

Liger0, 2020-12-01:
Hi, unfortunately after the second thread bumping I changed the configuration because with that one I couldn't print at all, and with the changes on the probe it works. I upped the Z to 10 and something else. I will share the working profile to make clearer what the fix was if you want to hunt the culprit.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5624921/Marlin.zip)

sjasonsmith, 2020-12-03:
@Liger0 if you are no longer encountering issues we can probably just close this issue.
From the description I have so far I think the behavior might actually be expected.

We should only keep this open if you believe there is a bug, can provide exact steps to reproduce, and are interested in spending time following troubleshooting steps to help isolate a cause.

Otherwise, we will just close this. If there is a real bug here somebody else will tell us soon 🙂

Liger0, 2020-12-03:
> @Liger0 if you are no longer encountering issues we can probably just close this issue.
> From the description I have so far I think the behavior might actually be expected.
>
> We should only keep this [...]

Well, the 0.10 error was  consistant unless I did reboot the machine, it wasn't a mesh value or anything like that and ruined the abl compensation as it didn't compensate for the bed correctly.
Now that the offset is 10, it doesn't go to 10.10 or anything like that.
The few edits I did in the configuration made the difference between triggering that behaviour or not.
If someone else has the free machine to try the behaviour as for the reproducing steps I posted he could do the investigation, at the moment my machine is working for the first time after years so until I do reupdate the machine I can't do much testing.

closed 2020-12-03 by Liger0