MarlinFirmware/Marlin #19415: [FR] Adding external buttons for Jogging Axis  X, Y and Z

yyazji, 2020-09-16:
# NO SUPPORT REQUESTS PLEASE

Do you want to ask a question? Are you looking for support? Please don't post here. Support Requests posted here will be automatically closed!

Instead use one of the following options:

- The Marlin Firmware forum at https://reprap.org/forum/list.php?415
- The MarlinFirmware Facebook Group at https://www.facebook.com/groups/1049718498464482/
- The MarlinFirmware Discord Server at https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the latest "bugfix" branch to see whether the issue is already addressed.

sjasonsmith, 2020-09-20:
There isn't enough information here to work with.

Jogging is already available through external displays, as well as through serial-connection controllers such as OctoPrint and Pronterface.

When I think about adding external buttons, I think I would need a LOT of external buttons to do anything useful, since I would need to be able to control multiple axis, jog at different rates, etc.

If you have ideas for a hardware design to make this useful, please provide more information in the description. If you do not already have a hardware design in mine, I suggest you look at the existing display and host software which can already proving jogging functionality.

closed 2020-09-20 by sjasonsmith

InsanityAutomation, 2020-09-21:
I would consider this a duplicate to the request to allow inputs to be fixed to macros which was a few manufacturers use for homing, or starting the most recent file on the sd card.

yyazji, 2020-09-21:
Thank you for your support!

actually, I am going to use this idea to control CNC machine not 3D
printer, Now it’s difficult to use software to jogging the axis and when I
jog the machine through Screen or LCD controller its move steps for example
10mm at turn rotate encode one step.
Recently, I saw some guys using joystick to control machine so I want to
make like joystick but through buttons

For example when pressing one button it moves continuously moving not step
move as found in lcd display

Thanks in advanced

*Yousef  M. Yazji | General Manager*
*AMEIC For Mechanical & Electrical Industries*
*Gaza - Palestine*
Phone:  +972 8   2840049
Mobile:  +972 59 8666177
Mobile:  +972 59 7909999
Skype: youyazji
E-mail: yyazji@gmail.com

sjasonsmith, 2020-09-21:
@InsanityAutomation you mean #7932?

It looks like there is a pull request open linked to that. @yyazji please look at that feature request and pull request to see if they help with your needs.

sjasonsmith, 2020-09-25:
@yyazji, you might look at this Pull Request to see if it would allow you to do what you want:
https://github.com/MarlinFirmware/Marlin/pull/18389

It should be pretty easy to apply these changes to your own branch and try it out. You experience with it could be useful as comments on that PR.