MarlinFirmware/Marlin #19417: LPC1769 SKR 1.4 Turbo MAX6675 compiles buy [BUG] (short description)

pediegoivan, 2020-09-16:
#19088 # Bug Description

I am building MARLIN 2.0.6.1 LPC1769 on SKR 1.4 Turbo.
I want to use the MAX6675 for my single extruder. I used to have an MKS SBASE with a MAX6675 running without any problem.

I defined the MAX6675 corresponding pins on the SKR 1.4 Pins file, then I declared "#define TEMP_SENSOR_0 -2"  in "configuration.h". The firmware compiles OK but, when the board flashes the firmware from the SD card both the SD Card stops being recognized by the pc and the board cannot communicate over USB serial -1 port (the only one that I currently have enabled).
If I define "#define TEMP_SENSOR_0 1" everything works fine again.
I tried defining different unused pins, for example EXP1 port
"""
#if !defined(MAX6675_SS_PIN)    // Thermocouple SS Pins
  #define MAX6675_SS_PIN   P1_20
  #define MAX6675_MISO_PIN P1_18
  #define MAX6675_SCK_PIN  P1_22
#endif
""""
The result is always the same. As soon as I define "#define TEMP_SENSOR_0 -2", the controller stops working.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5234851/Config.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

With an SKR 1.4 Turbo declare #define TEMP_SENSOR_0 -2" in "configuration.h" file

**Expected behavior:** [What you expect to happen]
Working MAX6675 for my extruder as it used to do it in MARLIN 2.0.5.3 in an MKS_SBASE board (LPC1768)

**Actual behavior:** [What actually happens]
The firmware compiles ok but when I flash it from the SD card, the SKR 1.4 Turbo ceases to work.

pediegoivan, 2020-09-16:
I have noticed that there are some example configuration.h files that call the "MAX6675_MISO_PIN" like "MAX6675_DO_PIN",
Therefore, I tried this, and now It won't compile. 
The error states: 
"""
//.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/module/temperature.cpp.o: in function `Temperature::init()':
//Marlin-2.0.6.1/Marlin\src\module/temperature.cpp:1810: undefined reference to `SPIclass<(unsigned char)9, (unsigned char)18, (unsigned char)7>::softSPI'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1
"""

rhapsodyv, 2020-09-16:
Please try the last bugfix-2.0.x. I tested with our configs and instructions and everything went fine.

pediegoivan, 2020-09-17:
I have just tried with the last bugfix-2.0.x
If I define "MAX6675_DO_PIN"  then:
The error states:
"""
//.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/module/temperature.cpp.o: in function Temperature::init()': //Marlin-2.0.6.1/Marlin\src\module/temperature.cpp:1810: undefined reference to SPIclass<(unsigned char)9, (unsigned char)18, (unsigned char)7>::softSPI'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1
"""
If I also "#define LPC_SOFTWARE_SPI" Same error.

If I only declare the "#define MAX6675_SS_PIN    P0_26" which should make use of the MOSI1 and SCK1 pins of the LPC1769. It does compile, but the communication with the MAX6675 still wont work and MAXTEMP triggers on power on.
The MAX6675 module was working until I disassembled the electronics this morning. I will test it again using the MKS SBASE that I removed and confirm that this is a software issue.
However, MAX6675 through SoftSPI should also work and it is currently not compiling.

pediegoivan, 2020-09-17:
I've tested my MAX6675 and it does not work anymore.
This morning was working. And it had its LED lit during all my tests. But now it does not turn on its LED anymore. I don't have any spare one to test again. I am going to assemble back my printer with the original E3D cartridge and if I have time later to get a new MAX6675 I will and repeat the test.
softSPI does not work dough but, should?

rhapsodyv, 2020-09-17:
> I've tested my MAX6675 and it does not work anymore.

It is not working anymore in the MKS SBASE too?

> This morning was working. And it had its LED lit during all my tests. But now it does not turn on its LED anymore. I don't have any spare one to test again. I am going to assemble back my printer [...]

It was working in the SKR 1.4 Turbo?

pediegoivan, 2020-09-17:
It never worked in the skr 1.4 turbo. It does not compile with softSPI.
It was working this morning with the mks sbase, it does not anymore

rhapsodyv, 2020-09-17:
> It was working this morning with the mks sbase, it does not anymore

Work: 2.0.5.3 in an MKS_SBASE board
Don't work: 2.0.6.1/bugfix? in an MKS_SBASE board
Don't work: 2.0.6.1/bugfix? in an SKR 1.4 turbo board

?

rhapsodyv, 2020-09-17:
I take a look in the actual code, and I very confident that it *never ever* worked with `MAX6675_DO_PIN`, because that `SPIclass<(unsigned char)54, (unsigned char)18, (unsigned char)54>::softSPI` is just declared and it is not defined in anywhere in the code, even in the 2.0.5.3.

Try this branch, I have fixed it: 
https://github.com/MarlinFirmware/Marlin/pull/19419

pediegoivan, 2020-09-17:
In 2.0.5.3 it did work. The pin was called MAX6675_MOSI_PIN.
I'm afraid that I wont be abme to try it just now. 
My max6675 did not turn on with the mks sbase. I fear that it might have burnt. I will confirm later if I make it work or get a new one.

rhapsodyv, 2020-09-17:
> In 2.0.5.3 it did work. The pin was called MAX6675_MOSI_PIN.

No, it wasn't called MAX6675_MOSI_PIN ... `MAX6675_MOSI_PIN` never existed! ... Defining it simply does nothing... So, I was right: you aren't using the softSPI in 2.0.5.3

If you want to try without the `MAX6675_DO_PIN`, you need to match the `MAX6675_DO_PIN, MOSI_PIN, MAX6675_SCK_PIN` with `MISO_PIN, MOSI_PIN, SCK_PIN`, and the SPI will work. With `LPC_SOFTWARE_SPI` or not (I guess).

rhapsodyv, 2020-09-18:
News?

pediegoivan, 2020-09-18:
No bud, sorry. I couldn't even check if my max6675 is working. if it isn't, I will get a new one and check this code next week

Grogyan, 2020-10-28:
Any news on this?

pediegoivan, 2020-10-28:
We can close this one. I couldn't try it. My max broke and I didn't have time time to get a new one install it, etc. Nobody else reported it so it must be my faulty board. I'm closing this

PR #19419 (2020-09-22): Definition for SoftSPI::softSPI MAX6675 
### Description

It maybe Fix #19417 .... 

But, as it's look like a clear C++ error (static variable without definition), I'm making this PR Ready for Review and Merge.

I think no one ever used MAX6675 with DO/MISO pin.... 

### Related Issues

#19417
closed 2020-10-28 by pediegoivan