MarlinFirmware/Marlin #19420: Bad print quality on MKS SBASE

keqinn, 2020-09-17:
### Bug Description

Hi everyone,

I get some trouble in printing with my MKS SBASE board: my 3D printer cannot print a fully vertical cuboid. Acturally, there always is a erro between each slices, so that the surface would be rough. The print result is shown in the below picture.

### My Configurations

I use Marlin2.0.5.4, and some configurations are changed to apply on my 3D printer. (Get my configuration files in )
One configuration change is that, I change the #define MINIMUM_STEPPER_PULSE 4 in configuration_adv.h, because my drivers cannot recognize too short pulse. The reason why I do not use the driver on MKS board is I need higher electric current to drive bigger motors.

Hope someone could help me, thanks in advance!

### Steps to Reproduce

1. download Marlin2.0.5.4
2. #define MINIMUM_STEPPER_PULSE 4 
3. print a cuboid

**Expected behavior:**The cuboid is vertical and its surface is smooth.

**Actual behavior:** The cuboid is crooked and its surface is rough.

#### Additional Information

I think there has no problem with my mechanical setting, because that I could print a perfect cuboid with Marlin1.1.9 and MKS GEN on the same printer.

* The reslut shows below.

[image]
[image]

ellensp, 2020-09-17:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

keqinn, 2020-09-18:
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Sorry, a new problem raised. I cannot upload bugfix-2.0.x branch into my MKS SBASE. My board will flash itself again and again, making me have no chance to connect with it.

ellensp, 2020-09-18:
Please attach your configuration files.

keqinn, 2020-09-18:
@ellensp 
[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5243955/configurations.zip)

sjasonsmith, 2020-09-20:
Some external drivers require extremely long pulse times. Do you actually have specifications for your hardware, or are you just guessing that 4ms is enough? Just driver specifications are not enough, because many external drivers use optocouplers with their own requirements.

The TB6560 is common in external drivers. When you configure Marlin to use `TB6560` it uses a 30 microsecond pulse time.
I suggest you configure all your drivers as `TB6560` and see whether that aids your print quality.

keqinn, 2020-09-25:
Sorry, I try it but the problem still exist. But I find a new feature: If I set the density to 0%, the printer will only print the shell, the surface get a little bit smooth( still have some layer shift). So maybe is an error caused by the movement when printing the interior part?

closed 2020-11-03 by github-actions[bot]