MarlinFirmware/Marlin #19425: [BUG] ABL doesn't work

Ducati87, 2020-09-17:
Hello, I don't know if it has already been discussed, but I'm using a Pinda2 for ABL but after the G29 there is no compensation. in some points it prints too squashed and in others too high and this implies that it is impossible to set the z offset. I'm using the ABL Bilinear actually. Is there any solution for this?

Thank you.

My config: SKR 1.4, TMC2209 and Pinda2 as probe.

BastR, 2020-09-17:
Du you use the last bug fix? I've this problem too I don't know if it's software or hardware for now

Ducati87, 2020-09-17:
I use the last stable release..

ellensp, 2020-09-18:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Ducati87, 2020-09-18:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate: [...]

Hello, thank you for your reply. Here you can find the log [https://pastebin.com/Xp1AdrDC](https://pastebin.com/Xp1AdrDC)

dubbc553, 2020-09-22:
I am also seeing this issue with the latest bugfix releases on my SKR Mini E3 1.2/ Ender 3 Pro

boelle, 2020-09-22:
@Ducati87 @BastR @dubbc553 

could you all attach your config files as a zip file?

Max05pl2, 2020-09-24:
> @Ducati87 @BastR @dubbc553
> 
> could you all attach your config files as a zip file?

I have the same issue, I put a bug but it got quickly removed for whatever reason.

dubbc553, 2020-09-26:
I think this may be fixed? Sorry, I was about to go through the log process and after flashing the latest firmware a few days ago the issue seemed to have gone away once I started a long print. Meant to respond sooner.

closed 2020-09-26 by thisiskeithb