MarlinFirmware/Marlin #19428: Refinement for Z_HOMING_HEIGHT

yysh12, 2020-09-17:
### Description
When my printer is first turned on using 2.0.6.1 and set to begin homing, the Z_HOMING_HEIGHT option ensures that the z-axis raises 4mm to ensure it doesn't scratch bed surface. However, when stepper motors timeout and power down, the z axis falls. Homing after it falls no longer raises z axis, presumably because it assumes z axis doesn't fall when powered off. This then leads to scratch on bed surface. A further refinement would be that the Z-axis always raises 4mm when it detects that the z-limit switch is activated to ensure it doesn't scratch bed. This also ensure that if at max height it doesn't raise beyond max limits as well because switch won't be active. Thus 3 zones of behavior: when z_min limit switch active: raise 4 mm before homing; when z unknown and min limit switch not active: don't raise; when z is known, below z_max, and limit switch not active: raise or don't raise depending on user preference. I should mention that the UNKNOWN_Z_NO_RAISE doesn't prevent bed scratch. 

### Feature Workflow

1. [First Action] Is limit switch active?
2. [Second Action] If yes, then raise 4mm before homing y and x axis.
3. [and so on...] 

#### Additional Information
I am using creality ender 3 
* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

qwewer0, 2020-09-17:
Does `HOME_AFTER_DEACTIVATE` helps?

yysh12, 2020-09-17:
Well, the problem is that even when homing it won't raise for clearance. it only appears to do that for the first home after powering on. So, when steppers deactivate, z axis falls, and homing is called on again, the extruder will scratch the surface and not raise 4mm. Which is why I was saying it should just always raise if limit switch is active.

qwewer0, 2020-09-18:
You are right, `HOME_AFTER_DEACTIVATE` does not rises the Z axis after the steppers have timed out.

sjasonsmith, 2020-09-20:
Please attach your Marlin configuration files, as requested by the issue template. Marlin behavior is extremely configurable, and it is very difficult to debug any issue without knowing how the firmware is configured.

yysh12, 2020-09-20:
Sorry for the late response, it was a busy week. But I edited the question to include config files with .h replaced with .txt so github won't complain.

yysh12, 2020-09-21:
So, while if the stepper motor drops to the plate, it will still continue to scratch the surface and refuse to raise up first, I found a solution that works for me. I set 
#define DISABLE_INACTIVE_Z false so that z never deactivates while on. 
Just thought I'd post that for any others with falling z axes.

yysh12, 2020-09-21:
Ideally it should still raise if Z_min is activated though

yysh12, 2020-09-30:
Okay so with some further experimenting, I have discovered the following:
If I manually disable the steppers from the LCD menu and then rehome:
The z axis falls after disabling and then it raises 10mm before homing, 
but if the stepper times out and the z axis falls, then it won't raise 10mm.

Why is this? what is the difference between a manual disable versus a timeout disable of stepper motors?

yysh12, 2020-10-06:
:-(

boelle, 2020-10-06:
as @sjasonsmith said

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sjasonsmith, 2020-10-06:
@boelle they already attached them to the issue description.

boelle, 2020-10-06:
must have been blind again... at least i did not see it attach 4 hours ago

qwewer0, 2020-10-06:
For me, after I homed the printer, then disabled the steppers via either M18, M84, LCD or timed out, the printer doesn't raise the Z axis for the next homing.

Step by step:

0. Turn on printer
1. Select Auto Home (first time)
2. Z axis rises by `Z_CLEARANCE_BETWEEN_PROBES` (I believe)
3. X and Y homed
4. X and Y moves to `Z_SAFE_HOMING`
5. Z axis rises by (`Z_CLEARANCE_DEPLOY_PROBE` - `Z_CLEARANCE_BETWEEN_PROBES` - Z probe offset) (I believe)
6. Z probed/homed
7. Z rises to `Z_CLEARANCE_DEPLOY_PROBE` (I believe)
8. Disable Steppers
9. Select Auto Home (second time)
10. X and Y homed
11. X and Y moves to `Z_SAFE_HOMING`
12. Z probed/homed
13. Z rises to `Z_CLEARANCE_DEPLOY_PROBE` (I believe)

I think this wasn't always the case, and the second homing should act just as the first one.

yysh12, 2020-10-07:
thanks guys :-) . Also yes I can second qwewer0's post. Mine follows that same sequence of events as well.

yysh12, 2020-10-19:
So I managed to fix my printer up so that it won't fall anymore, but I still think this is abnormal behavior.

qwewer0, 2020-10-19:
@yysh12 Could you please reopen this issue? There is clearly something off with the homing height, so this is still relevant, even if not for a bug, but for a feature request.

yysh12, 2020-10-19:
Oh sure no problem. I reopened it.

qwewer0, 2020-11-28:
Still a problem.

ellensp, 2020-11-28:
@qwewer0 with current bugfix or some older version?

yysh12, 2020-11-28:
Yes still with bugfix

yysh12, 2020-11-28:
Oh wait sorry with 2.7.2
Haven't tried it with latest bugfix

qwewer0, 2020-11-28:
@ellensp Yes, with the latest bugfix-2.0.x

The same thing happens as in https://github.com/MarlinFirmware/Marlin/issues/19428#issuecomment-704372728

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5610058/Configuration.zip)

thinkyhead, 2020-11-28:
If `UNKNOWN_Z_NO_RAISE` is enabled there will be no raise at the start of `G28` unless the Z axis has been homed and the Z steppers have remained powered all the time since the last `G28`. Please disable `UNKNOWN_Z_NO_RAISE` if you have a bed or Z gantry that stays in place when powered off, and you don't care about Z being raised after the steppers have turned off.

qwewer0, 2020-11-28:
@thinkyhead I don't have `UNKNOWN_Z_NO_RAISE` enabled. And neither @yysh12 have it enabled in the provided config zip.
I think my z axis stays in place fairly good, but I wouldn't trust it, so a raise would be nice after the steppers have been turned off.

qwewer0, 2020-11-28:
> Okay so with some further experimenting, I have discovered the following:
> If I manually disable the steppers from the LCD menu and then rehome:
> The z axis falls after disabling and then it raises 10mm before homing,
> but if the stepper times out and the z axis falls, then it won't raise 10mm.

@yysh12 When I disable the steppers from the LCD, the rehoming does not raises the z axis.

yysh12, 2020-11-28:
Oh interesting. Yours won't raise at all. Yeah mine does raise on manual disable just not auto disable. Disabling inactive z didn't seem to help either.

yysh12, 2020-11-28:
Oh wait. I didn't read it correct. You have the z homing height disabled. You have to have that enabled for it to raise.

yysh12, 2020-11-28:
For me the problem is that for some reason auto timeout of steppers behaves differently than manual disable or steppers, such that z homing height is ignored on auto timeout.

qwewer0, 2020-11-28:
> Oh wait. I didn't read it correct. You have the z homing height disabled. You have to have that enabled for it to raise.

I think after M84 or stepper timeout the next homing should behave as it behaves when first homed, because after the printer can no longer know where each axis is, then it should act as on the first homing.

New information: If I move the Z axis from the LCD or serial then after I disable the steppers, the the next (first) homing acts as the problematic homing, so without any z axis rise, as if it still knows the Z axis position, despite the fact that the Z axis could have sunk.
So it seems, that M84 or stepper timeout doesn't erases the last Z position.

thinkyhead, 2020-11-30:
You can try reverting this change from 73ce80af3a3, which was done to address some other issue…
```diff
- do_z_clearance(z_homing_height, TEST(axis_known_position, Z_AXIS), DISABLED(UNKNOWN_Z_NO_RAISE));
+ do_z_clearance(z_homing_height, true, DISABLED(UNKNOWN_Z_NO_RAISE));
```

qwewer0, 2020-11-30:
On the latest bugfix-2.0.x (presumably because of https://github.com/MarlinFirmware/Marlin/pull/20323)
This is what happens compared to https://github.com/MarlinFirmware/Marlin/issues/19428#issuecomment-704372728:
Now between the 9th and 10th point:

- Z axis rises by `Z_CLEARANCE_BETWEEN_PROBES` (I believe)

Which is a big leap in the positive direction, as a z axis drop after stepper deactivation is no longer a problem (unless I'm missing something), because the Z axis rises on the next homing, but still, after the 10th point it is not rises the Z axis by (`Z_CLEARANCE_DEPLOY_PROBE` - `Z_CLEARANCE_BETWEEN_PROBES` - `Z probe offset`) (I believe)

@thinkyhead Could you confirm one thing for me or point my to a code?
After the XY axis is homed and moved to `Z_SAFE_HOMING` position, where or how much the Z axis will rise? (5th point on https://github.com/MarlinFirmware/Marlin/issues/19428#issuecomment-704372728) By my crude measurements, it will rise an additional (`Z_CLEARANCE_DEPLOY_PROBE` - `Z_CLEARANCE_BETWEEN_PROBES` - `Z probe offset`), but I'm just assuming this.

Thank you for https://github.com/MarlinFirmware/Marlin/pull/20323, and in general.

swissnorp, 2020-11-30:
Sorry guys but i dont think this is a bug... try to enable the following in configuration_adv.h:
```
// If the Nozzle or Bed falls when the Z stepper is disabled, set its resting position here.
#define Z_AFTER_DEACTIVATE Z_HOME_POS
```

Mentioning the following pull request / commit:
https://github.com/MarlinFirmware/Marlin/pull/18906

**We may should consider to move this setting to configuration.h as we did with following pull request for `HOME_AFTER_DEACTIVATE`:**
https://github.com/MarlinFirmware/Marlin/pull/20283

yysh12, 2020-12-01:
Hello, so I actually am away from my printer until March 10th so I will be unable to test anything out until then. That being said, I will definitely try out your recommendations once I get back home. But I should also note that I actually already tried utilizing that option to no avail. The main problem I am trying to understand is why my automatic stepper timeout is behaving differently than my manual stepper timeout. Basically if I manually disable the steppers, Z_homing_height works as expected and provides clearance from the build plate, but if the steppers auto-timeout they refuse to provide the clearance. And both tests were done at the same height. But I will definitely take a closer look at the Z_after_deactivate next time I with my printer and Ill have to let you know then.

swissnorp, 2020-12-11:
@yysh12 Ok, thank you for your reply.

@qwewer0 What about you? Could you test this?

qwewer0, 2020-12-11:
@swissnorp Sorry, didn't saw your edit.

First of all, this is the first homing sequence on my printer:
(which I believe should be the homing sequence after a stepper deactivation)

1. Z axis rises by `Z_CLEARANCE_BETWEEN_PROBES` (I think)
2. X and Y homed
3. X and Y moves to `Z_SAFE_HOMING`
4. Z axis rises by (`Z_CLEARANCE_DEPLOY_PROBE` - `Z_CLEARANCE_BETWEEN_PROBES` - `Z probe offset`) (I think)
5. Z probed/homed
6. Z rises to `Z_CLEARANCE_DEPLOY_PROBE` (I think)

Now on to the testing:

- Without any extra feature enabled:
If the next homing after a stepper deactivation (from LCD) is called when the z position is lower than `Z_HOMING_HEIGHT` (maybe?), then it behaves just like the first homing, otherwise if the z position isn't lowered, then it homes without the 4th point (above), which is now my main concern.
- With `Z_AFTER_DEACTIVATE` (and therefore `Z_HOMING_HEIGHT` = `Z_CLEARANCE_BETWEEN_PROBES`):
After a stepper deactivation (from LCD), if the next homing is done from lower, higher or from the same z position as the previous homing ended, it behaves just like the fist homing without any extra movement anywhere.

swissnorp, 2020-12-11:
@qwewer0 
So you can confiirm it's not a bug, right?
It behafes as it should as mentioned?

> Without `Z_AFTER_DEACTIVATE` the Z-axis will only rise if you moved it below `Z_HOMING_HEIGHT` before releasing the stepper and it will only rise by the difference between `z.current_position` and `Z_HOMING_HEIGHT`

I will open a pull request to move `Z_AFTER_DEACTIVATE` to configuration.h.

qwewer0, 2020-12-11:
> @qwewer0
> So you can confiirm it's not a bug, right?
> It behafes as it should as mentioned?

As I mentioned it, I think the next homing after a stepper deactivation should behave the same as the first homing, but it doesn't! And I don't think it should need a z axis is lowering to do so.

So, I still think that there is a bug as it does not behave as it should.

swissnorp, 2020-12-12:
> * Without any extra feature enabled:
>   If the next homing after a stepper deactivation (from LCD) is called when the z position is lower than `Z_HOMING_HEIGHT` (maybe?), then it behaves just like [...]

Thats exactly how it should behave!
`do_z_clearance` is not intended to raise everytime you call it!

https://github.com/MarlinFirmware/Marlin/pull/18576
https://github.com/MarlinFirmware/Marlin/pull/18906/files

> As I mentioned it, I think the next homing after a stepper deactivation should behave the same as the first homing, but it doesn't! And I don't think it should need a z axis is lowering to do so.
> 
> So, I still think that there is a bug as it does not behave as it should.

If you want this to happen you must enable `Z_AFTER_DEACTIVATE`. Otherwise marlin assumes that z is hight enough and clearance is available.

qwewer0, 2020-12-12:
> Otherwise marlin assumes that z is hight enough and clearance is available.

That is (one of?) the main problem. On a freshly bought printer, `Z_AFTER_DEACTIVATE` most likely isn't enabled, as the manufacturer doesn't think that their Z axis would fall, as on their testing it did not, but if with time the Z-rod runs smoother and/or it is greased, then the Z axis might fall at some point without the user knowledge, and not everyone able or willing to thinker with marlin, and that is why marlin should always assume that after a Z stepper deactivation the Z height isn't useable/trustable, and so it should always raise it on the next homing.
At the moment on the latest bugfix-2.0.x, the homing after Z stepper deactivation is almost perfect, to the point that I would almost say that there is no problem anymore, but that isn't the truth. (https://github.com/MarlinFirmware/Marlin/issues/19428#issuecomment-735587576)

closed 2021-01-19 by github-actions[bot]

neilvangeffen, 2021-05-14:
Im having a similar issue, but i get absolutely no z raise, regardless of what i try, and im stumped....
Even when i turn the machine on first time, connect to simplify3d and press home all, or send G28, it just homes X, they Y and then Z, with no Z Raise anywhere... It's been driving me bananas for the last 4 hours, and i cant find the problem

Was using 2.0.7.2 and that didnt work, so then moved to latest bugfix 2.0 and also no dice

[Cant get Z to Raize.zip](https://github.com/MarlinFirmware/Marlin/files/6478762/Cant.get.Z.to.Raize.zip)