MarlinFirmware/Marlin #19430: [BUG] Home Fail On creality 4.2.7 Without SD card inserted

theonefromthesky, 2020-09-18:
Creality 4.2.7 Ender 5 

Fresh copiled binary works fine but will only home successfully on x axis with the SD card inserted. 

Without card in printer axis will ram in to gantry and completely ignore end stop, with sd card inserted and reboot, homing works as intended.

thisiskeithb, 2020-09-18:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Please follow the Issue Template when filing a bug report. This includes attaching your configs:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

thisiskeithb, 2020-09-18:
I'm confident that pulling the latest `bugfix-2.0.x` should fix this issue since it was reported in https://github.com/MarlinFirmware/Marlin/issues/19388 and was resolved in PR https://github.com/MarlinFirmware/Marlin/pull/19226.

thisiskeithb, 2020-09-18:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19388

closed 2020-09-18 by thisiskeithb