MarlinFirmware/Marlin #19433: Dual z axis endstops and bltouch - z offset ignored

Maxinger15, 2020-09-18:
Hello,
I am using 2 z endstops on my printer for homing and a bltouch for bedleveling. After homing the nozzle is about 10mm above the printbed. I tryed to fix this with the z offset of the bltouch but this doesn't change anything. I disabled the software min z endstop but even then no change. How can I configure Marlin to use the z probe offset for the normal endstops?

boelle, 2020-09-18:
read the template !!

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Maxinger15, 2020-09-18:
[Marlin_2_0_6_1_conf.zip](https://github.com/MarlinFirmware/Marlin/files/5246630/Marlin_2_0_6_1_conf.zip)
 Sorry for the trouble. Here are my configs. I will try the bugfix branch too.

Maxinger15, 2020-09-18:
The problem is still here with the current version in the bugfix-2.0.x branch.

shitcreek, 2020-09-18:
**

> Are you looking for support? Please don't post here. Instead please use one of the support links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

**

Have you tried adjusting these settings?
`#define Z_AFTER_HOMING  2      // (mm) Height to move to after homing Z

`#define Z_AFTER_PROBING           10 // Z position after probing is done`

Maxinger15, 2020-09-29:
For me z offset fixed it. But I think it would be nicer if it can be done via the z probe offset if a bltouch is attached.

closed 2020-09-29 by Maxinger15