MarlinFirmware/Marlin #19437: BLtouch does not probe entire bed when leveling

DaveP-Hawaii, 2020-09-19:
Ender 3 Pro, Bigtreetech SKR Mini E3 V2 board, Creality BLtouch with its metal mounting bracket, Marlin V2.0.6.1 with Mini E3 V2 Config files copied to it. Am running a Micro Swiss Direct Drive hotend, but that shouldn't matter.

I've gone into the Marlin software and reconfigured it for the BL touch. Set the offsets from the nozzle and defined the corners. I had to set the probing margin to 60 just to get the first probing point on the bed. When bed leveling is preformed, the initial homing is not in the center of the bed, but in the front left quadrant. Than bed leveling starts in the front left corner, moves over X by about 53 mm for second probe, than 53 mm more for third probe which is about the center of the bed, than move back on Y and continues the 3 by 3 pattern, but not going to back of bed on the Y. I've  played with the numbers, but cannot get it to change. I've tried changing the margin the 70, but that just made the grid probing smaller in size.  ????

Thank you for any help.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5248768/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5248769/Configuration_adv.zip)

thisiskeithb, 2020-09-19:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-19 by thisiskeithb

DaveP-Hawaii, 2020-09-20:
I believe this is a possible bug report. Marlin V2.0.6.1 was just released in August. I've attempted to locate anybody having the same issues with the new version with no success. The only issues that have been addressed, either in instructional or problems, that I've found, are with older versions of Marlin and other motherboards. Perhaps it's in BigTreeTech SKR Mini E3 2.0's config files that come in the Configurations-import-2.0.x folder for Marlin-2.0.x.

When the original Creality board V4.2.2 was in the printer with Creality's BLtouchs firmware, an older version of Marlin (V1.2 or something of the sort), the bed probing worked correctly.

The only way that I have found to make the probing grid larger is by changing the default X and Y bed size of 235 to 300 for X and Y. With this change loaded to my hardware, the BLtouch probes the entire bed, with the nozzle staying over the bed. This appears to be a work a round to the issue, but is it the correct way to address it?

Of course I may just be lost in the weeds on this.

thisiskeithb, 2020-09-20:
> I believe this is a possible bug report.

It's not.

> Perhaps it's in BigTreeTech SKR Mini E3 2.0's config files that come in the Configurations-import-2.0.x folder for Marlin-2.0.x.

I built that config and have personally run it on an Ender-3 with an SKR Mini E3 V2 and BLTouch (multiple versions) and it probes as expected.

> The only way that I have found to make the probing grid larger is by changing the default X and Y bed size of 235 to 300 for X and Y. 

Your `PROBING_MARGIN` is set to 60 which leaves you with 115mm<sup>2</sup> of probing area on a 235mm<sup>2</sup> bed.

---
Again: 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).