MarlinFirmware/Marlin #19438: [BUG] or 2 with WS2812B Neopixels

cr0wm4n, 2020-09-19:
2  Led issues with Marlin 2.0.6.1

I have connected 60 WS2812B Neopixel type LED Strip to my Trigorilla board, I am using a servo pin for them.
The LED's are GRB 5V type and the LED's work fine as an ambilight on my TV with 260 in the strip
I am using a 10A 5V buck converter to drive them which is connected directly to the PSU. 
There is a common ground between buck converter, mainboard and the LED's

When turning the printer on they perform the LED test cycle fine. When starting to heat up the bed the Leds sequentially light up but every so often will make a mistake, the last lit led will go out and all the other leds will change to semi random colors. then when the next led lights up the colors go correct again.

after the bed has finished heating and the hotend starts to heat up its almost like a disco, the leds all randomly change colors. 

once the hotend reaches temperature the Led's all go white fine.
I would expect them to work as described in configuration.h

The 2nd part of the issue
While testing the LED's if I send the command M150 U50 The leds will go green but if I send the same command again, then they go red or blue, then send again the leds turn off.
It will do the same sort of thing if i send M150 R50 or M150 B50 

I would expect the colors to not change when sending the same command 2 or 3 times in succession.

ellensp, 2020-09-19:
Please attach your configuration files.

cr0wm4n, 2020-09-19:
I thought i had attached them, it refused the raw confs but accepted the zip, Attaching again
[Marlin_config_files.zip](https://github.com/MarlinFirmware/Marlin/files/5249003/Marlin_config_files.zip)

ellensp, 2020-09-19:
Your using a ATmega2560 based controller, slow 8 bit..

You cannot control a large number of neopixels on this controller reliably while running marlin.

Neopixels requires precise timing, marlin requires precise timing.  One of these has to give and it wont be marlin.

closed 2020-09-19 by ellensp