MarlinFirmware/Marlin #19439: Load Filament via Change filament Menu on idle machine

mikemerryguy, 2020-09-19:
In 2.0.5 I could have an idle machine with NO filament loaded
I have load and unload gcodes on
advanced pause on
park is on
Turn fresh machine on and click Change Filament Menu
Load Filament
Preheat PLA
it would preheat to PLA preheat temp
BEEP and wait for me to click the button THEN load the filament.

Now in 2.0.6.x 
Skip ahead a little, I click 
Preheat PLA
it preheats, beeps once quick, then starts loading automatically, then starts purge all automatically

No wait for USER to click before extruder starts the load and purge / purge more-continue routine.

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-09-19:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

mikemerryguy, 2020-09-19:
@ellensp this is not a USER SUPPORT question
It is a MARLIN BUG
the OLD Firmware worked as intended, 2.0.6.1 DOES NOT
it beeps and starts loading filament without user interaction as it is supposed to do!!!

thinkyhead, 2020-10-05:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2020-10-05:
@thinkyhead This was opened again in https://github.com/MarlinFirmware/Marlin/issues/19455

thisiskeithb, 2020-10-05:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19455

closed 2020-10-05 by thisiskeithb