MarlinFirmware/Marlin #19440: [BUG] (platformio build chain broken for tronxy chitu boards)

tuskiomi, 2020-09-19:
### Bug Description

When using platformio, the corresponding firware.bin and firmware.elf files are incompatible with the chitu board's processor

### My Configurations

--Valid, tested configurations for the chitu board

### Steps to Reproduce

1. configure your marlin to be compatible with the chitu boards from tronXY.
2. use platformio's build feature as set up from the marlin website to build binaries for the configured marlin.
3. upload the binaries using STM32Cube programmer, and verify the integrity of the upload
4. restart the board
**Expected behavior:** a marlin splash screen is seen on the LCD, followed by the main menu

**Actual behavior:** No screen is shown, and the board will not update from SD card's update.cbd files. no noise is made, no behavior is present.

#### Additional Information

A hex dump of my firmware.bin can be found at: 

https://pastebin.com/FHMKvihc

ellensp, 2020-09-19:
When the firmware is generated the last step is to encrypt the bin as this is what the bootloader expects.
If you are bypassing the bootloader by uploading directly with STM32Cube then it is not going to work.
You need to edit  buildroot/share/PlatformIO/scripts/chitu_crypt.py so it doesn't do the encryption.

closed 2020-09-19 by ellensp

tuskiomi, 2020-09-19:
@ellensp 

I'm uploading at the bootloader, that's what the example instructions say to do.. Commenting out the extra script line results in a build failure because of a file not found error:

```
c:/users/brice/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file chitu_f103.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\chitu_f103\firmware.elf] Error 1
```

tuskiomi, 2020-09-19:
Note that when the line is uncommented, the build is done successfully

rhapsodyv, 2020-09-19:
You messed up with the bootloader. You need put back the chitu firmware to restore the bootloader and just copy the update.cdb to sd card.

The current chitu env is not intended to use with direct stm 32 cube programmer. 

Using the current method, copying update.cdb to SD, is far easier and simple than using stm 32 cyber programmer.

tuskiomi, 2020-09-19:
@rhapsodyv I did everything according to the xyguide [here](https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Tronxy/X5SA/HOWTO-INSTALL.md), I'm not sure that I strayed from the instructions.

tuskiomi, 2020-09-19:
also @rhapsodyv, I would prefer an arduino or STM bootloader in lieu of the chitu bootloader anyway.

rhapsodyv, 2020-09-19:
> @rhapsodyv I did everything according to the xyguide [here](https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Tronxy/X5SA/HOWTO-INSTALL.md), I'm not sure that I strayed from the instructions.

The guide mark the "Flashing Marlin Firmware MANUALLY" as "(OBSOLETE!)"...

tuskiomi, 2020-09-19:
Note that the start of the file has the following hex:

```
00f0 0020 a18b 0008 9196 0008 9596 0008
9996 0008 9d96 0008 a196 0008 039e 0008
039e 0008 039e 0008 039e 0008 039e 0008
039e 0008 039e 0008 039e 0008 159e 0008
039e 0008 039e 0008 039e 0008 039e 0008
039e 0008 039e 0008 e596 0008 0597 0008
2597 0008 4597 0008 6597 0008 3995 0008
5195 0008 6995 0008 8595 0008 a195 0008
bd95 0008 d995 0008 fd92 0008 039e 0008
39aa 0008 039e 0008 039e 0008 8597 0008
49a1 0008 6da1 0008 91a1 0008 c9a1 0008
1da2 0008 8da2 0008 fda2 0008 ed9a 0008
059b 0008 f99a 0008 119b 0008 039e 0008
039e 0008 a5a7 0008 f5a7 0008 45a8 0008
c997 0008 039e 0008 039e 0008 25a4 0008
49a4 0008 6da4 0008 a5a4 0008 5593 0008
039e 0008 039e 0008 6da3 0008 039e 0008
95a8 0008 e5a8 0008 dda3 0008 01a4 0008
f595 0008 0d96 0008 2596 0008 4196 0008
```

which looks a lot like a series of ARM jumps, indicative of the start of memory, where ISR jumps could be made.

tuskiomi, 2020-09-19:
> The guide mark the "Flashing Marlin Firmware MANUALLY" as "(OBSOLETE!)"...

normally, I wouldn't use such a method, but when your board won't take update.cbd s anymore, one must fall back to old methods.

rhapsodyv, 2020-09-19:
> normally, I wouldn't use such a method, but when your board won't take update.cbd s anymore, one must fall back to old methods.

The better approach is to put Chitu back, to restore bootloader.

All 3 files: cdb, bin, elf, get encrypted, so they are incompatible with direct flash. You need to comment the line 122 of `chitu_crypt.py` to disable the encrypt process, to use it. But it don't worth. Better is to restore bootloader and use update.cdb.

tuskiomi, 2020-09-19:
@rhapsodyv Chitu cannot be put back once the bootloader is written over.

Also, the output does not change whether or not the line is commented or not, as checked by the diffchecker.

rhapsodyv, 2020-09-19:
> @rhapsodyv Chitu cannot be put back once the bootloader is written over.

You surely can. Just put back your chitu backup using the STM32 cube programmer. If you backup right, as in the tutorial, you are able to flash it.

tuskiomi, 2020-09-19:
@rhapsodyv  the backup does not include the bootloader, and is encrypted.

rhapsodyv, 2020-09-19:
> @rhapsodyv the backup is a gcode file, and is not any firmware.

Hum... I wrote that tutorial... There're two sections: 
"Save your printer settings", for chitu code: https://github.com/rhapsodyv/Configurations/blob/tronxy_x5sa_v5_v6/config/examples/Tronxy/X5SA/HOWTO-INSTALL.md#save-printer-settings-optional-but-recommended

And, "Backup your chitu firmware":
https://github.com/rhapsodyv/Configurations/blob/tronxy_x5sa_v5_v6/config/examples/Tronxy/X5SA/HOWTO-INSTALL.md#backup-your-chitu-firmare-optional-but-strongly-recommended

tuskiomi, 2020-09-20:
@rhapsodyv  I see, I didn't back up my firmware.

rhapsodyv, 2020-09-20:
Try asking in the Tronxy Facebook Group. Someone will have it to share with you.