MarlinFirmware/Marlin #19448: [FR] Ability to abort SD print with the intention of resuming

logiclrd, 2020-09-19:
### Description

A couple of times I've had prints fail partway through and have wanted to resume them from where they failed. Marlin didn't know that they had failed at the time, and so I find the printer moving around doing nothing in mid-air above the print. By closely inspecting the layers that were printed, I can often tell exactly what layer it stopped at (using gyroid infill means that, if nothing else, the infill is slightly different on every layer). So, I abort the print from SD card, then open up the G code on my computer, delete everything up to that layer, sort out whatever the issue was and re-prime the nozzle, then tell Marlin to print the partial file. But, it doesn't line up perfectly with the print, and I think the cause maybe mostly comes down to two things:

1. This is hypothetical, and I would be glad to find out that this not the case :-) But, when the print is stopped, it calls `Planner::quick_stop`, and I see in the code for that that it effectively erases whatever buffered actions haven't yet been delivered, by setting `block_buffer_head` equal to `block_buffer_tail`. Is it possible that it might skip a change in Z and then the printer no longer accurately knows its location?

2. When the steppers stop moving, I think after a while they get disabled, which I think means that the axes, and Z in particular, are prone to drifting. So, the print head might not be at the height it thinks it is at when you come back to it some time later.

In addition to these things, stopping the print also immediately shuts off the bed heater and the hot end. I had an attempt to continue a print fail a few weeks back because I neglected to immediately reverse the change in bed heat, and so the incomplete part popped off the bed a few minutes after I began working to recover it.

I'd like a feature in the Marlin UI, and maybe G code, that allows me to tell the printer I want it to stop the current print _with the intent that it'll be picked up, possibly from a different G code file_, so that the bed doesn't cool down, and the steppers are kept active to maintain positional accuracy and stuff.

### Feature Workflow

A menu item similar to "Pause Print" and "Abort Print" would act as sort of halfway between the two, an "abort" in the sense that it gives up on the current G code, but a "pause" in the sense that it _doesn't_ give up on the current print. Perhaps `Begin Recovery` or `Recovery Needed` or something.

qwewer0, 2021-01-01:
https://github.com/MarlinFirmware/Marlin/issues/16970 Duplicate, one or the other can be closed.

thisiskeithb, 2022-03-13:
Duplicate of #16970

closed 2022-03-13 by thisiskeithb