MarlinFirmware/Marlin #19450: [BUG] TMC2209 + SKR 1.4 Turbo Stealthchop cannot be disabled

pediegoivan, 2020-09-20:
### Bug Description

If I disable stealthchop commenting out the lines :
"""
  #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  #define STEALTHCHOP_E
"""
(disabling all the additionally needed lines for the code to compile)

I get a different result from selecting a very small Hybrid Threshold value for each axis (M913 X1 Y1 Z1 E1)
I also cannot manage to get a smooth enough transition between stealthchop and spreadcycle. even at 15 mm/s. If I do a move with a speed that matches the threshold value the steppers become noisy or I hear periodic knocks with the movement, even with TMC_DEBUG and MONITOR_DRIVER_STATUS  both disabled.

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5250944/Config.zip)

### Steps to Reproduce

Enable stealthchop and move at threshold speed (any value selected for threshold speed results in vibration)

Selecting threshold to be a very small value i.e. 1mm/s with any move being a spreadcycle move is different from disabling the stealthchop in firmware.

**Expected behavior:** [What you expect to happen]
Be able to get a smooth transition between stealthchop and spreadcycle at some speed (at least one)
or alternatively be able to disable stealthchop and always use spreadcycle.

**Actual behavior:** [What actually happens]

only way to use spreadcycle was to select a very small hybrid threshold value in firmware and/or M913.

Quas7, 2020-09-21:
Just to rule out something simpler (no offense) ;)
Did you also reset your eeprom settings after flashing the FW?  e.g. M502 followed by M500 or re-init eeprom?
Normally, Marlin should ask for re-init of the eeprom during startup but not sure what triggers this.
Or you did set #define EEPROM_AUTO_INIT  anyways.

pediegoivan, 2020-09-21:
I dont understand how would that influence it. I disabled the stealthchop in firmware and it resulted in being locked in stealthchop with m913 not being recognized.
The only way i have found to run spreadcycle always was to enable stealthchop and hybrid threshold and to set the threshold at a very low value. "M913 X1 Y1 Z1 E1"

pediegoivan, 2020-09-21:
> Just to rule out something simpler (no offense) ;)
> Did you also reset your eeprom settings after flashing the FW? e.g. M502 followed by M500 or re-init eeprom?
> Normally, Marlin should ask for [...]

I did not re-initialize the EEPROM as it was not requested by the firmware

Quas7, 2020-09-21:
Ok, now I got it. You disabled stealth chop in the FW but it stays active for you (without re-initializing eeprom, if it matters at all). 
And with the command M569 S0 X Y Z E you are also not able to disable stealth chop as intended either? https://marlinfw.org/docs/gcode/M569.html

At least for me, the M569 does change the mode as expected.

sjasonsmith, 2020-09-21:
> I did not re-initialize the EEPROM as it was not requested by the firmware

It will only display the message if the EEPROM layout changed, usually if you changed firmware versions. If you only changed settings it will happily ignore the settings in your header and use stale values from EEPROM.

Quas7, 2020-09-21:
Quickly tested now. I can confirm some interessting behavior.

Disabled X,Y,Z stealth chop in FW. 
 // #define STEALTHCHOP_XY
 // #define STEALTHCHOP_Z
 // #define STEALTHCHOP_E
I did not change anything else - compiled directly as expected.

Interesstingly, M569 still confirms stealthChop set (command is not disabled nor reporting spreadCycle instead).

Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	stealthChop
Recv: Z driver mode:	stealthChop

Resetting the eerpom with M502 to defaults changes the values to spreadCycle.

Send: M569
Recv: X driver mode:	spreadCycle
Recv: Y driver mode:	spreadCycle
Recv: Z driver mode:	spreadCycle

And more important, I can still switch on/off stealth chop with M569...

Send: M569 S1 X Y Z
Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	stealthChop
Recv: Z driver mode:	stealthChop

And it also has an effect as it directly changes the sound of my printer very characteristcally.

On the other hand, M913 is not recognized anymore.

Send: M913
Recv: echo:Unknown command: "M913"

I would conclude, that swithching off stealthChop in firmware does not fully deactivates it, if previously set in eeprom or manually via M569.

Therefore, a possible workaround for the time being is using M569 S0 X Y Z E and store it with M500 instead of disabling it in FW. Resetting eeprom works as well.

pediegoivan, 2020-09-21:
Hahahaha, I was about to report the same. Yes! The disabling stealthchop in the firmware just thinks that disabled it, but M569 still works and M913 does not. Exactly what you said.
Thanks for your help with regards of M569. I did not knew that that gcode existed. I now have my extruder with Hybrid threshold working (the only axis in which that works reliably wtf) and fixed spreadcycle in all my other axis. Now I am able to reach speeds of 600mm/s for travel moves. My in stealthchop mode I am able to reach 150mm/s reliably but, if I set the HYBRID_THRESHOLD to 170mm/s I still cannot reach the 600mm/s that I am able to in fixed spreadcycle mode. I would be great to have spreadcycle for only travel moves, as these are the ones that might reach the hybrid threshold value more frequently...
I have just found a [FR] that asks exactly for this! #13316.

pediegoivan, 2020-09-21:
However, we agree that there is a [BUG] confirmed here.
Disabling stealthchop in firmware should be able to fix all the drivers in spreadcycle and disable M569 and M913.

ManuelMcLure, 2020-09-22:
> However, we agree that there is a [BUG] confirmed here.
> Disabling stealthchop in firmware should be able to fix all the drivers in spreadcycle and disable M569 and M913.

That is not correct. Disabling StealthChop in firmware sets the _default_ to SpreadCycle, but should never lock the drivers to that value. It's the same as any other value that's stored in EEPROM - the `Configuration_adv.h` value specifies the default which can be overriden by gcode and stored again in EEPROM. It can be argued that the names of the parameters could be better (maybe `STEALTHCHOP_DEFAULT_XY` instead of `STEALTHCHOP_XY`?)

pediegoivan, 2020-09-22:
Ok... Agreed. That would be waaaay clearer. Or adding a comment that states "override with M569" around there too.

phazei, 2020-09-24:
@pediegoivan do you think any of the issues you're having related to this at all?
https://github.com/MarlinFirmware/Marlin/issues/19050

When changing the firmware between hybrid or stealthchop, it seems it needs to be de-powered, not just rebooted for it to take affect.  I ran across that issue quite a bit while troubleshooting.  To be able to test hybrid easily and not need to reflash each time, I've just set the threshold to 0 for everything which performs identically to having hybrid threshold disabled, but it still leaves the menu options so it can be adjusted for testing.  I'm not able to just leave the threshold 1 because then sensorless homing breaks, and I have no endstops.

pediegoivan, 2020-09-24:
That does not happen to me. If I change hybrid threshold values with M913 I hear the difference in motors vibration in different speeds without power loop

jagoosw, 2020-10-18:
Which version of marlin are you using? I don't have this problem with the exact same setup

Quas7, 2020-10-18:
For my same setup, the bug seems only to manifest, if the eeprom is not initiated after flashing back and forth.

closed 2020-11-25 by github-actions[bot]