MarlinFirmware/Marlin #19453: [BUG] SKR 1.4 Filament Runout Sensor on Pins 1.26 & 1.25 don't work when printing M600 is not executed

kkberg, 2020-09-20:
### Bug Description

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5252047/Config.zip)

### Steps to Reproduce

1. try sensors with M119 when filament is in and out
2. [start a print
3. simulate an empty filament spool

**Expected behavior:** [What you expect to happen] I would expect running of M600 which is configured

**Actual behavior:** [What actually happens] No reaction on missing filament M412 is activated

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

kkberg, 2020-09-22:
Switched to a SKR 1.3 with same firmware and same issue.

martinziener, 2020-09-22:
I have the same issue on skr1.4.
M412 is activated, M119 shows correct working; M600 not activated, just keeps printing

thisiskeithb, 2020-09-22:
I just tested a couple SKR 1.4 builds running `bugfix-2.0.x` (225487f9db7d1a7aaf6886a283c7758a9ae2181e) with a filament runout sensor connected to `FIL_RUNOUT_PIN`/`P1_26` and filament runout detection works as expected in both cases:

- Begin print
- Cut filament
- Runout is triggered & `M600` is kicked off
- Filament automatically unloads
- Prompted to load filament/press encoder to continue
- Filament is loaded
- Print continues after clicking Continue on the Purge More screen

`M119` output with filament present: `filament: TRIGGERED`
`M119` output with filament removed: `filament: open`

Have your filament runout sensors worked before? Can both of you test the latest `bugfix-2.0.x` to see if this is still an issue?

kkberg, 2020-09-23:
Thanks for your reply. My sensors worked before. M119 works well and also when printing M119 status is correct but M600 isn't triggered. M600 is activated. I'll try the bugfix later today.
Thank you again.

kkberg, 2020-09-23:
I tried it and M600 isn't triggered. Used the latest bugfix from today.  I use these switches, they show right status with M119 even when printing and cutting filament. 
[image]

kkberg, 2020-09-23:
> 
> 
> I just tested a couple SKR 1.4 builds running `bugfix-2.0.x` ([225487f](https://github.com/MarlinFirmware/Marlin/commit/225487f9db7d1a7aaf6886a283c7758a9ae2181e)) with a filament runout sensor connected to `FIL_RUNOUT_PIN`/`P1_26` and filament runout detection works as expected in both cases:
> 
>     * Begin print
> 
>     * Cut filament
> 
>     * Runout is triggered & `M600` is kicked off
> 
>     * Filament automatically unloads
> 
>     * Prompted to load filament/press encoder to continue
> 
>     * Filament is loaded
> 
>     * Print continues after clicking Continue on the Purge More screen
> 
> 
> `M119` output with filament present: `filament: TRIGGERED`
> `M119` output with filament removed: `filament: open`
> 
> Have your filament runout sensors worked before? Can both of you test the latest `bugfix-2.0.x` to see if this is still an issue?

Which kind of switches do you use Keith?

thisiskeithb, 2020-09-23:
> Which kind of switches do you use Keith?

A couple of Prusa's latest IR filament sensors and a basic 3 pin/switch style.

kkberg, 2020-09-23:
I tried a 3 pin switch without pcb and also not working. But all switches shop right M119 values. In the moment I've no idea what is causing the problem. Maybe the other guys who have same problem try out additional things. But first thank you for your help.

martinziener, 2020-09-23:
Running the latest bugfix from this morning didn't help me neither; problem remains

thisiskeithb, 2020-09-23:
> The only difference I see is: M119 output with filament present: filament: OPEN

It should say `TRIGGERED` when filament is present. Change `FIL_RUNOUT_STATE` to `LOW`.

martinziener, 2020-09-23:
Changed FIL_RUNOUT_STATE to LOW; M119 output with filament present: filament: TRIGGERED
But M600 is still not triggered in any way

thisiskeithb, 2020-09-23:
Do you get any output over serial? Also, please zip up and attach your configs for review.

martinziener, 2020-09-23:
No, no output over serial 
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5269126/config.zip)

kkberg, 2020-09-23:
I already did tests with LOW or HIGH before writing the issue. Also not triggering M600. It shouldn't matter if it's configured LOW or High, it should trigger if status of switch changes. Keith please provide your config to find any other settings which could influence. Thank you

medline2c, 2020-09-23:
Bonjour, même soucis avec SKR 1.4 Turbo et capteur optique !... IMPOSSIBLE de lancer une impression : il me dit toujours qu'il n'y a plus de filament !

* AVEC le filament :
M109 donne : TIGGERED 
M412 donne :
Filament runout ON

* SANS le filament :
M109 donne : OPEN
M412 donne :
Filament runout ON également

Mes paramètres (Configuration.h) :
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  // #define FILAMENT_RUNOUT_DISTANCE_MM 5

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    //#define FILAMENT_MOTION_SENSOR
  #endif
#endif

dans pins_BTT_SKR_V14.4.h :
//
// Filament Runout Sensor
//
#define FIL_RUNOUT_PIN                     P1_26  // E0DET    => J'ai bien connecté le S du capteur sur cette pin
#define FIL_RUNOUT2_PIN                    P1_25  // E1DET

J'utilise Marlin 2.0.6.1...

thisiskeithb, 2020-09-23:
Here's an example config that uses a simple switch: [SKR14-example.zip](https://github.com/MarlinFirmware/Marlin/files/5269243/SKR14-example.zip)

> PS what do you mean with output over serial? Or is a TFT meant.

I mean like terminal output over USB/in your host software like Pronterface/OctoPrint/etc.

thisiskeithb, 2020-09-23:
@medline2c Please don't paste your config since translating your comments will be more difficult. Just zip them up and attach.

medline2c, 2020-09-23:
> @ medline2c Veuillez ne pas coller votre configuration car la traduction de vos commentaires sera plus difficile. Fermez-les simplement et attachez-les.

Désolé, je suis nouveau ici, j'ai supprimé les commentaires pour laisser l'essentiel : je dois tout supprimer ou je laisse ainsi ? 
J'ai regardé l'exemple  SKR14-example.zip, je ne vois rien de différent de ma configuration si ce n'est HIGH au lieu de LOW pour FIL_RUNOUT_STATE. J'ai essayé HIGH et ça donne le même résultat..

medline2c, 2020-09-23:
> @ medline2c Veuillez ne pas coller votre configuration car la traduction de vos commentaires sera plus difficile. Fermez-les simplement et attachez-les.

J'ai testé tes fichiers avec MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO

Aucun changement, mêmes symptômes que décrit précédemment ...

kkberg, 2020-09-23:
I tried Keith's config and it also doesn't work on my SKR 1.4 . I tried several switches and had no luck.

thisiskeithb, 2020-09-23:
Check your wiring as well?

kkberg, 2020-09-23:
Wiring is okay, V G S are right, also when using a 2 wire switch. Otherwise there could be a short. M119 shows right status.

martinziener, 2020-09-23:
Same here; Everything looks and registers ok but marlin doesn't seem to notice the pin change.

martinziener, 2020-09-23:
Success! Changing #define FILAMENT_RUNOUT_SCRIPT "M600" to "...M600 T0" did the trick for me.
(got it from issue #18341)

Shouldn't this be documented somewhere within Marlin?

thisiskeithb, 2020-09-23:
> Success! Changing #define FILAMENT_RUNOUT_SCRIPT "M600" to "M600 T0" did the trick for me.
> (got it from issue #18341)

`M600` should work with the active extruder.

@martinziener: you shouldn't have had to specify the extruder since since your config is for a single extruder/hotend. `M600` works without `T0` on my single nozzle/single extruder builds, so it's odd that it doesn't work the same for yours.

In the case of @kkberg's config with dual extruders, specifying `M600 T0` may have an undesired effect of performing the filament swap on the incorrect extruder.

martinziener, 2020-09-23:
It works with T0, tried again without T0 but no success.

thisiskeithb, 2020-09-23:
Just for good measure, are you Initializing EEPROM after each flash? You can do it through the LCD or by sending `M502` followed by `M500`.

martinziener, 2020-09-23:
:-) yes

kkberg, 2020-09-23:
Tried M600 T0 but still same problem. I now try it out with defining only on sensor.

kkberg, 2020-09-23:
No succesess, I now go down to one extruder and try out if it depends on extruder count.

kkberg, 2020-09-23:
still not working ... @martinziener would you please send me your config and commit version you use? Thank you

martinziener, 2020-09-23:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5271539/config.zip)
It's the bugfix version from this morning, I don't know where I can find the exact commit version

kkberg, 2020-09-23:
thank you, I'll make the needed changes for my printer. I also use that version.

medline2c, 2020-09-23:
Je ne peux tester cette solution car ma version ne me permets pas de démarrer l'impression : il me dit toujours qu'il n'y a pas de filament... dois-je ouvrir un autre post pour ce soucis ou est-ce lié ?

martinziener, 2020-09-23:
@medline2c; I don't think these problems are related

kkberg, 2020-09-24:
Didn't work with one extruder too. Don't know if it makes sence to go on with the sensors. Printing is possible without sensors. I think if it works on several boards and on my board not it could be hardware but I'm not convinced. May be I give Klipper a try or Smoothieware.

kkberg, 2020-09-24:
Just got a smart runout sensor and it also doesn't work on the 1.4 . Game over ... with BTT

martinziener, 2020-09-24:
Which drivers are you using? Are they using the Diag piin, in that case it wouldn't work

kkberg, 2020-09-24:
My drivers are TMC2130 but don't have the diag pin soldered, I don't use sensorless homing. That was the first thing I did and took the pins off.

martinziener, 2020-09-24:
You could try the P0_10 probe pin, at least that one doesn't have hardware pullups

kkberg, 2020-09-24:
I'll give it a try, thank you

Trenchi, 2020-09-24:
Hello Guys, 

"First: Sorry for my bad English. I'm from Germany and had a little help from a translator."

I have this problem too. I use a SKR 1.4 Turbo with TMC2208 a Bigtreetech 2 in 1 Out hotend and 2 runout sensors on a TFT 35 V3.0. (Marlin V2.0.6.1)

Both runout sensors are connected to pin 1.25 and 1.26. 

At pin 1.26 a signal bridge is connected to the TFT 35 to trigger the runout sensor in touchscreen mode.  (If you don't do this, the sensor only works in touchscreenmode and not in marlin mode of the display but that's not important here)

As I said, Runout Sensor 1 of pin 1.26 on the board is additionally connected to the TFT 35 (Only Ground & Signal). 

If I now print with the TFT 35 in touchscreen mode the Runout Sensor 1 works (it is bridged to the display). If I now print in Marlin mode for Filament Mixing then BOTH runout sensors do NOT work!

With Pronterface I can trigger both runout sensors easily and they work.

I also think it is a Marlin problem because the sensor works on the TFT35 display and this has his own firmware.

Today after work I will try the current bugfix version. Maybe the problem is solved.

thisiskeithb, 2020-09-24:
Please attach your configs. Also: Have these sensors worked in a prior version of Marlin?

It's odd that there's now a run of people having issues all of sudden, especially since I can't reproduce them on my boards.

Trenchi, 2020-09-24:
yes, everything worked in V2.0.6.0 

I noticed it when I changed to the Bigtreetech 2in 1out a few days ago.  There I updated to V2.0.6.1.

after the rebuild i only set the Mixing in the configuration files.

The sensors are also only gimmicks. It also works without. But as it is, if it is there it should work.

I attach my Config V2.0.6.2 with Mixing and also my old Config V2.0.6.0 WITHOUT Mixing. 

With V2.0.6.1 there was also the problem that my Z-axis went in the wrong direction (I have CoreXY) and so I had to change the direction of rotation.

Yesterday I played a little bit with VSCode and I hope that no strange things are activated in the current configuration. I tried to solve the problem with #define FILAMENT_RUNOUT_SCRIPT "M600" to "M600 T0". But I didn't flash it to the printer because there were a lot of yellow warning messages when compiling.

I am still at work and can only show the configuration that is stored on the NAS. So if you look at my configuration V2.0.6.1 please ignore everything that is unnecessary and has nothing to do with runout sensors.
[Config v2.0.6.1 Mixing E.zip](https://github.com/MarlinFirmware/Marlin/files/5275983/Config.v2.0.6.1.Mixing.E.zip)
[Config V2.0.6.0 Without Mixing.zip](https://github.com/MarlinFirmware/Marlin/files/5275984/Config.V2.0.6.0.Without.Mixing.zip)

thisiskeithb, 2020-09-24:
@Trenchi Please try with the latest `bugfix-2.0.x` to see if your runout sensors work.

thisiskeithb, 2020-09-24:
I pinged @rhapsodyv since there were some possible PRs of his that may have affected this board and he's received similar reports from users with Chitu boards (`STM32F103RE`).

This _could be_ a deeper Marlin issue, but he's looking into it.

kkberg, 2020-09-24:
Just tried P0.10 and also correct values with M119 but not triggering M600. @Trenchi  I'm from Germany too :)

kkberg, 2020-09-24:
> 
> 
> I pinged @rhapsodyv since there were some possible PRs of his that may have affected this board and he's received similar reports from users with Chitu boards (`STM32F103RE`).
> 
> This _could be_ a deeper Marlin issue, but he's looking into it.
@thisiskeithb  That sounds promissing and hopefully he will find it. I already prepared a lot of boards with Marlin and I had always luck :). Thank you Keith for your help

rhapsodyv, 2020-09-24:
@kkberg 

Did the printer pause or it keep working? 

Did you try changing `FILAMENT_RUNOUT_SCRIPT` to `M125` or `M25` to test?

I'm trying to isolate the problem.

kkberg, 2020-09-24:
@rhapsodyv print keeps printing without any pause. I didn't try M125 or M25 because I didn't know that it's possible to use any other gcode. I'll try it soon. Thank you very much for your fast answer.

kkberg, 2020-09-24:
@rhapsodyv tried M125 instead of M600 but also same behavior. I mad a little video but only have to hands to manage :). 
https://youtu.be/bRG2cztqjiA

rhapsodyv, 2020-09-24:
@kkberg do you have two sensor? Both aren't working?

kkberg, 2020-09-24:
> 
> 
> @kkberg do you have two sensor? Both aren't working?

Yes, but now I test with only one sensor to make it easier.

rhapsodyv, 2020-09-24:
Lets do a test. In the file `Marlin/src/feature/runout.h` line 44, uncomment it:
```
//#define FILAMENT_RUNOUT_SENSOR_DEBUG
```

Run a test and past here the serial output.

kkberg, 2020-09-24:
> 
> 
> Lets do a test. In the file `Marlin/src/feature/runout.h` line 44, uncomment it:
> 
> ```
> //#define FILAMENT_RUNOUT_SENSOR_DEBUG
> ```
> 
> Run a test and past here the serial output.

I'm just compiling

Trenchi, 2020-09-24:
@kkberg Hab ich mir schon fast gedacht, frag mich nicht warum :-)

i found something out.

if #define MIXING_EXTRUDER is enabled then the runout sensors do not work.

if I disabled #define MIXING_EXTRUDER then the sensors work again.

should it be that way?

kkberg, 2020-09-24:
I tried several times, but no output when sensor ist triggered. Seems that runout.h is not touched.

kkberg, 2020-09-24:
> 
> 
> @kkberg Hab ich mir schon fast gedacht, frag mich nicht warum :-)
> 
> i found something out.
> 
> if #define MIXING_EXTRUDER is enabled then the runout sensors do not work.
> 
> if I disabled #define MIXING_EXTRUDER then the sensors work again.
> 
> should it be that way?

Bist Du eventuell in unserer FB-Gruppe https://www.facebook.com/groups/364846444121852/requests/?hoist_pending_member_ids_suggested_order=100003731024132&notif_id=1600955952536028&notif_t=group_r2j für SKR Boards

Trenchi, 2020-09-24:
@kkberg Nee leider nicht, ich bin nicht so aktiv bei Facebook....ich hab meinen Account seit 5 Jahren nicht mehr geupdatet ^^

kkberg, 2020-09-24:
> 
> 
> @kkberg Nee leider nicht, ich bin nicht so aktiv bei Facebook....ich hab meinen Account seit 5 Jahren nicht mehr geupdatet ^^

@Trenchi hätte ja sein können.

[50 more replies]

closed 2020-10-02 by thisiskeithb