MarlinFirmware/Marlin #19454: [BUG] Comment for assisted tramming does not match points that need to be set

Mike8-I, 2020-09-20:
### Bug Description

The comment in Configuration_adv.h how to set up the tramming points seems not to be correct.
" // Define positions for probing points, use the hotend as reference not the sensor."
When setting up the points as commented, the tramming does not complete (fails at point 1 (counting from 0).
I took the assumption, that the probe position needs to be the set-point, contrary to the comment and that #define PROBING_MARGIN , from bed leveling needs to be obeyed, too. Also the extruder must not leave the min and max bed size, taking the probe to nozzle offset into consideration. To simplify things I defined the printer as a one extruder type instead of two.

Working configuration: 
The extruder now goes to x=189  and y=187 maximum when tramming (as expected if my assumptions is correct)
#define X_BED_SIZE 195
#define Y_BED_SIZE 194
NOZZLE_TO_PROBE_OFFSET { -64, -16, 0 }
PROBING_MARGIN 6
// Define positions for probing points, use the hotend as reference not the sensor.
  #define TRAMMING_POINT_XY { {  6, 7 }, { 125,  7 }, { 125, 171   }, { 6, 171 } }

I also suspect that the number of screw turns might not be calculated correctly, as the screw position needs to be known on the bed and relative to the tramming points to calculate correctly.

### My Configurations
[Configuration Tramming.zip](https://github.com/MarlinFirmware/Marlin/files/5252026/Configuration.Tramming.zip)

**Expected behavior:** setup beeing succesfull when beeing set up as commented
**Actual behavior:** setup is successfull when set up opposite to the comment

#### Additional Information
#Error compiling in 2.0.6 due to G35.cpp and ASSISTED_TRAMMING with BLTouch present #19402

sjasonsmith, 2020-09-20:
I haven’t looked at this code, but it probably assumes your probe points are exactly over the screws.

Vertabreak, 2020-09-20:
something i noted in testing this feature was that if your offset prevents reaching the set point it will fail to function correctly for me my offset is -40 so i had it do 40 away from all edges and it works correctly.

here is a video of it working on one of my printers : https://www.youtube.com/watch?v=EQABwXEO688

Yenya, 2020-09-23:
Just a me too. Banging my head against the wall over the "G35 failed at point ..." error message, I have discovered that the comment in Configuration.h is wrong - the TRAMMING_POINT_XY is indeed given in the probe as reference, not hot-end.

That said, I would prefer the code to be modified to match the comment, not vice versa.

Yenya, 2020-10-26:
Ping

closed 2020-12-05 by github-actions[bot]

marndra, 2021-01-01:
Please reopen it, it's still an issue. The Info "use the hotend as reference not the sensor" is not correct!

sjasonsmith, 2021-01-01:
> Please reopen it, it's still an issue. The Info "use the hotend as reference not the sensor" is not correct!

@marndra the text `use the hotend as reference not the sensor` has already been removed. It was confusing, so I removed it with my earlier change that validates the tramming points at build-time.

marndra, 2021-01-01:
Hi,

I used configuration.h from the geeetech example out of the Beta 2.0x from
today 01.01.2021.
Here it is still there. But i will check it tomorrow at the marlin default
config.

Good to know, thanks!

Q
mit freundlichen Grüßen
Martin Haßlöcher

sjasonsmith, 2021-01-02:
There is a bit more process involved for the change to propagate to the examples. It should end up there eventually.

marndra, 2021-01-02:
Hi Jason,

 

at first, many thanks for the info and the earlier fix.

I’ve seen you changed the Text for the trammming in the cong adv, geat, also in all examples, woow.

 

But I got a compilation error now because it seems you removed today the “ASSISTED_TRAMMING_MENU_ITEM”.

 

Why this?

 

Regards Martin

qwewer0, 2021-01-02:
> But I got a compilation error now because it seems you removed today the “ASSISTED_TRAMMING_MENU_ITEM”. Why this?

`ASSISTED_TRAMMING_MENU_ITEM` was removed because it cannot be used with only just the LCD, as that was just a button for G35 gcode command, which will only send the results through serial connection (Pronterface, Octoprint, etc.).