MarlinFirmware/Marlin #19455: Load Filament Issue

mikemerryguy, 2020-09-20:
### Bug Description

Load Filament not waiting for user input.
### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5252237/configs.zip)
](url)

### Steps to Reproduce

1. Turn on machine, have no filament loaded
2. Click through menu to LOAD FILAMENT / PREHEAT "x"
3. Wait

**Expected behavior:** [What you expect to happen]

In older Marlin 2.0.5 the Machine would preheat to set temp of Preheat X (label 1 or 2) usually PLA /ABS or PLA /PETG
The machine would hit target temp, beep and WAIT for user to click to start feeding filament.
It will then purge and display Purge more or continue.

**Actual behavior:** [What actually happens]

In 2.0.6.1 The Machine will Preheat and then give a quick BLIP of a beep and start loading filament by itself, then purge by itself and finally display Purge more or Continue.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

https://www.dropbox.com/s/3vzqk4jvnam7xrh/Expected%20%28oldmarlin%29.mp4?dl=0
https://www.dropbox.com/s/8zyqs6sjy9f8do9/New%20Marlin%20BUG.mp4?dl=0

This isn't a User Cofig error either !!
This is a Marlin BUG
the menu_filament.cpp file has a lot of changes and I can't track down the one causing issue.

It skips the wait for user as you can see in the videos!!!

qwewer0, 2020-09-20:
I have the same issue. (Ender 3, SKR Mini E3 v1.2)
https://github.com/MarlinFirmware/Marlin/issues/18191 might has the same root of problem.

thisiskeithb, 2020-09-21:
I can confirm that filament load/unload no longer waits for user input, but I'm not sure if it's an intentional behavior change or a bug.

I'll try to do some digging and find the commit that changes this behavior.

qwewer0, 2020-09-21:
@thisiskeithb It's a bug or an unintentional change, because we had and should still have the option to to purge more on filament load.
I will test the marlin before https://github.com/MarlinFirmware/Marlin/pull/18738, to see if it is any different.

thisiskeithb, 2020-09-21:
I have the option to purge on my builds when loading filament with the latest `bugfix-2.0.x`, so that behavior doesn't appear to have changed.

qwewer0, 2020-09-21:
I couldn't compile https://github.com/MarlinFirmware/Marlin/commit/42fbd527f3bcc317ae6851aca5a45bdbf979de55, so I can't test the version before https://github.com/MarlinFirmware/Marlin/pull/18738.
But one for sure, that I don't have the ability to purge more or any in filament load. After clicking filament load it heats to the target nozzle temperature, then after a beep, it loads the filament then just exits back to the main screen.

<details>
  <summary>42fbd52 Error message</summary>
  
ModuleNotFoundError: No module named 'platformio.managers.package':
  File "C:\users\user\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(item, exports="env")
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\Marlin-42fbd527f3bcc317ae6851aca5a45bdbf979de55\buildroot\share\PlatformIO\scripts\common-features-dependencies.py", line 12:
    from platformio.managers.package import PackageManager
  
</details>

thisiskeithb, 2020-09-21:
@qwewer0: The original post says they have the option to purge more or continue and I can’t reproduce what you’re seeing when loading filament since I also am prompted to purge more or continue when using the Load Filament option with the latest `bugfix-2.0.x`.

qwewer0, 2020-09-21:
@thisiskeithb Sorry, I overlooked that.
From the original post I only have matching the "start loading filament by itself", but then my problem could be from a different origin.

qwewer0, 2020-09-21:
@thisiskeithb For some reason I had `ADVANCED_PAUSE_CONTINUOUS_PURGE` enabled, so with that disabled, my experiences are now matches with @mikemerryguy's

mikemerryguy, 2020-09-22:
Thanks @thisiskeithb  Keith, another guy brought it up on FB and I was trying to add the feature myself and figured I should test old marlin first, and sure enough it worked as intended!!!

thisiskeithb, 2020-09-22:
"As desired" may be a better phrase. I still can't confirm if this is a bug or an intended behavior change, but it is behaving different than older versions of Marlin.

qwewer0, 2020-09-22:
If this is an intended behavior change, then I'm ok with it, as it is easy to have the filament ready in the extruder when I start the filament load.

mikemerryguy, 2020-09-23:
@thisiskeithb @qwewer0 I am pretty sure it isn't INTENDED as the machine starts the BEEP phase like older marlin but the beep is cut short and it skips straight to loading and extruder moving in FAST LOAD mode....

really I don't think its intended, and I really think its bad if so...

Think of a Direct Drive user, that can't just pop the filament in past gear as it would be too far or cause issues!!!

or Dual extruders, can't have them all just start spinning free as they want!!!!

I for sure think this is a bug, as I stated menu_filament.cpp has MAJOR changes vs 2.0.5 
and I for one think the issue is in there... I just did a text compare of 
m600
pause.cpp
menu_filament.cpp

most were just fixing coding and cleaning that up a bit...
menu_filiament has a TON of changes!!!

thisiskeithb, 2020-09-23:
> Think of a Direct Drive user, that can't just pop the filament in past gear as it would be too far or cause issues!!!

I have bowden & direct driver printers and have tested these changes and they don't break anything. It's just a change in behavior at this point.

mikemerryguy, 2020-09-23:
@thisiskeithb 
It's not desired 
I know it "works" but it isn't right...

I unloaded after print
Power down come back and click load 
Expect my new machine to do exactly as every other machine in my farm..

But it messes up and I have to babysit or do special steps for THIS ONE

Again.. donit and watch and listen close 

The beep is interrupted and skips to loading

I guarantee it is an error!!

mikemerryguy, 2020-09-23:
@thisiskeithb Keith
I have been playing for a bit since I got home..
looking at code and messing with changes...
Then I decided to try 
Idle Machine
Change Filament
Preheat PLA
it preheats and Unloads as normal
then gets stuck in a Loop for heating nozzle and will never pass to Insert Filament and then Loading Filament and then onto the purge more or continue loop!!!!

Going to dig into that area next!!

mikemerryguy, 2020-09-23:
@thisiskeithb @qwewer0 

I have taken IDLE machine several times and click change filament menu, CHANGE filament, it get stuck and I just keep clicking PREHEAT NOZZLE 
If I walk away I get Thermal Runaway Error....

Please try to Change filament from idle machine!!!

I have dug though some changes on this issue as well with same results on menu_filament.,cpp !!!

lots of changes in this file and I have reverted and tweaked a lot down the file, no success yet!!! still same on LOAD from idle and CHANGE from idle!!!

thisiskeithb, 2020-09-24:
> I have taken IDLE machine several times and click change filament menu, CHANGE filament, it get stuck and I just keep clicking PREHEAT NOZZLE
If I walk away I get Thermal Runaway Error....

> Please try to Change filament from idle machine!!!

I have and my printers complete the change without error.

qwewer0, 2020-09-24:
Change Filament does the following for me:
Select Change Filament -> I select temp. -> Heats to the selected temp. -> Unloads the filament -> Beeps a few times ->
Waits for user -> I click for it to continue -> Loads filament -> Waits for user to purge more or to continue (both works) ->
I click continue -> Change Filament is completed, and heating is still on. (There was no Z movement if that mattered)

So it looks good for me.

mikemerryguy, 2020-09-24:
that is strange, everything is stock except a few changes in .h and adv.h, ill re download and try again, I still can't get the regular load from idle to work proper, tried several changes in menu_filament.cpp (went back to stock before trying this so I thought, maybe I missed something)  Ill pull latest bugfix and recompile later... chicken paprikash for dinner sooooo not messing around tonight LMAO

mikemerryguy, 2020-09-28:
Just downloaded all new from 1 hour ago update on bugfix

Change filament works as intended that must have been my error.

Load Filament weather nozzle is cool or at set temp, still gives a blip and starts to load without waiting for user input.

Any insight from developers if this is INTENDED or BUG

I am still saying bug... 
as the machine blips (small beep versus intended BEEPS set in firmware for user notification) then just starts to load on its own!

Thanks guys.

thisiskeithb, 2020-09-28:
> Any insight from developers if this is INTENDED or BUG

I've been busy with some other commitments, but I should have time to test some older versions of Marlin this week to track down when this changed and why.

mikemerryguy, 2020-09-28:
@thisiskeithb Thank you sir!
No Rush, just keeping the "bug" active so it doesn't auto close!!!

benklop, 2020-09-29:
I'm seeing similar behavior, but I have a dual extruder machine. AFAICT, no matter which extruder I select, it operates on the first extruder. It also calls them extruder 1 and extruder 2, which is slightly confusing since they are E0 and E1 elsewhere.

NightFalcon650, 2020-09-29:
I've have a Vertex K8400. The filament load and unload menu has one flaw.
It doesn't matter what extruder i select it always selects the 2nd extruder. Why is this and where can i change this?

In the old marlin you could easily say where on the xyz-axes the head has to go for filament load en unload. How is this exactly arranged in the new marlin?

benklop, 2020-09-29:
@NightFalcon650 thank you for the tip - i was beginning to wonder how i was going to change out the filament in extruder 1 if i could only ever get 2 to load/unload. this certainly seems poorly tested, but at least if there are workarounds I can use the machine.

NightFalcon650, 2020-09-29:
> @NightFalcon650 thank you for the tip - i was beginning to wonder how i was going to change out the filament in extruder 1 if i could only ever get 2 to load/unload. this certainly seems poorly tested, but at least if there are workarounds I can use the machine.

I just started to inspect the code (also not a proffesional programmer) but i get the idea that the problem lies in the fact that the load/unload/replace is derived from the ultimaker, due to the tools changing part etc.. Somewhere in the code the heads arent selected wright, you can also see that on the menu. Extruder 1 is the right one, but on the display it's the left one. things are switched. Maybe someone could solve this issue.

mikemerryguy, 2020-10-05:
Still same, just commenting for "auto close" issue!

thisiskeithb, 2020-10-05:
You only need to comment when the bot adds the stale label.

mikemerryguy, 2020-10-06:
ah, didn't know, will you get an email for bot stale label like you do comments?

working on a new G code feature... once I get it working I'll submit a feature request with new and edited files!

its very raw right now and not working, the programming is making me mad bc I am self taught and not great HAHA
get things working but takes me more time than it would you!!!

thisiskeithb, 2020-10-06:
> will you get an email for bot stale label like you do comments

Yes, because it leaves a comment on the issue.

thinkyhead, 2020-10-14:
These kinds of things are always tricky, so some ability to do trial and error with your own setup(s) always helps.

mikemerryguy, 2020-11-14:
Have not updated mailin in a while issue still persists on original version. Very annoying
Can't even dona change filament as it auto loads and starts to purge all by itself

rhapsodyv, 2020-11-21:
I just found the cause and fixed it: #20243

Anyone can test it?

Thanks

qwewer0, 2020-11-21:
@rhapsodyv It looks like https://github.com/MarlinFirmware/Marlin/pull/20243 fixed the issue. Great job!

rhapsodyv, 2020-11-21:
It was merged. I will close this. Thanks.

PR #20243 (2020-11-21): Fix Load Filament Wait for User
### Description

PR #17315 did a lot of changes for "wait for user", but remove one `wait_for_user = true` that should not. This is clear typo, as right bellow the code have a while(wait_for_user).

This PR fix that, putting wait_for_user back where it was.

### Benefits

Fix #19455 

### Related Issues

#19455 
#17315
closed 2020-11-21 by rhapsodyv