MarlinFirmware/Marlin #19457: [BUG] (SKR 1.4 onboard SD card not working)

feiticeir0, 2020-09-20:
### Bug Description

Onboard SD card is not recognized and only works for the board do firmware upgrades. I'm using the TFT35 v2.0 from BigTreeTech that also has an SD card. This one works. 

If I try to read the SD card from the SKR 1.4 board, I get an error:
"Unknown command M21" 

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Press print
2. Choose onboard SD
3. Error

**Expected behavior:** [What you expect to happen]
It's expetect to list the files on the SDcard and enable to print 

**Actual behavior:** [What actually happens]
Error 
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5252710/Configurations.zip)
[image]
[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-09-20:
You don't have `SDSUPPORT` enabled in your Configuration.h file.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-20 by sjasonsmith

feiticeir0, 2020-09-20:
I'm so sorry !
I did search for that and I really really thought I had it selected. I'm going to try now ! 
Thank you