MarlinFirmware/Marlin #19463: [BUG] Cannot compile with ADVANCED_PAUSE_FEATURE without PREVENT_COLD_EXTRUSION

sjasonsmith, 2020-09-21:
### Bug Description

If `ADVANCED_PAUSE_FEATURE` is enabled, you cannot compile with `PREVENT_COLD_EXTRUSION` disabled, due to the following error:

```
Marlin\src\feature\pause.cpp: In function 'bool ensure_safe_temperature(bool, PauseMode)':
Marlin\src\feature\pause.cpp:132:51: error: 'class Temperature' has no member named 'extrude_min_temp'
     thermalManager.setTargetHotend(thermalManager.extrude_min_temp, active_extruder);
```

I believe this was introduced by a recent change which added some heating logic when resuming from pause. I wasn't quite sure how to resolve this build failure, without possibly undoing the intentions of the prior change.
828a582f4dfc719dbb519b89cccf621a5184d2a1

I frequently use `PREVENT_COLD_EXTRUSIONS` when reproducing bug reports. It allows me to reproduce problems with heaters disabled and no filament loaded.

### My Configurations

These are the default configurations, with minimal changes to reproduce the build failure.
[Configurations_COLD_EXTRUSION.zip](https://github.com/MarlinFirmware/Marlin/files/5253438/Configurations_COLD_EXTRUSION.zip)

### Steps to Reproduce

1. Build.

#### Additional Information

N/A

rlojek, 2020-10-13:
Confirmed, both bugfix and 2.7.x branches.

rlojek, 2020-10-13:
I did dome  research, in [temerature.h](https://github.com/MarlinFirmware/Marlin/blob/8e1ea6a2fa1b90a58b4257eec9fbc2923adda680/Marlin/src/module/temperature.h#L320) field extrude_min_temp is declared conditionally in respect to **PREVENT_COLD_EXTRUSION**. This was not changed recently, yet how the **ensure_safe_temperature** handles this has changed:
[2.6.1](https://github.com/MarlinFirmware/Marlin/blob/ca194ca52ee63fe319305a79e396b8b013b4c935/Marlin/src/feature/pause.cpp#L122) no use of field
2.0.7 and 2.0.x use that field:
[https://github.com/MarlinFirmware/Marlin/blob/da79674f84ef65cb014288a27d60f8709b1f0936/Marlin/src/feature/pause.cpp#L136](url)

sjasonsmith, 2020-10-26:
Thanks for the fix, @rlojek!

closed 2020-10-26 by sjasonsmith