MarlinFirmware/Marlin #19466: Clean install of Arduino v1.8.13 can not build current BugFix

Roxy-3D, 2020-09-21:
### Bug Description

After doing a clean install of Arduino v1.8.13 the latest BugFix can not be built.    I did the install and modified my sketch location to be C:\tmp\a    just to insure file name length would not be a problem.    I also edited the preferences.txt file to include build.path=C:\Tmp\a to keep the file name lengths passed to GCC at a value that Windows can tolerate.

### My Configurations

I used Arduino's library manager and included the latest version of the U8glib library. (version 1.19.1)

### Steps to Reproduce

Telling Arudino to build the code fails.    It can not find the U8glib.h header file.  The error message is below:

sketch\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp:65:10: fatal error: U8glib.h: No such file or directory
 #include <U8glib.h>
          ^~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

Is anybody else having problems with the latest BugFix with Arduino?

thisiskeithb, 2020-09-21:
Configs please 🙂

Did you try with Arduino IDE 1.9?

Roxy-3D, 2020-09-21:
> Did you try with Arduino IDE 1.9?

No.   Not yet.   But for sure we want Marlin to compile with the 'Current Release' of Arduino.

I'm wondering if some code in the AVR HAL got checked and never tested against the Arduino build system.
This is a very strange error.   Removing the U8glib library and reinstalling that does not help either.

Roxy-3D, 2020-09-21:
Trying Bob's suggestion from here:  https://github.com/MarlinFirmware/Marlin/issues/9422#issuecomment-362337783

ellensp, 2020-09-22:
This is a windows limitation. there is nothing marlin can do about it.
Windows has a hard coded 32k character limit on its fork/exec function.
When arduino IDE links marlin together it makes one enormous line linking all generated .o files together. 
This is the wrong way to do it. 

This is why we say platformio is preferred It does things intelligently. only complies what is needed produces smaller binaries and smartly links without upsetting windows.

Shortening the path to Marlin helps a little.
Shortening the build path helps a little.
Removing unused HAL directories helps a little
Removing unused UI helps a little. 

But these are all just band-aids on a badly leaking dam that is the arduino ide.
As Marlin grows, these tricks will work less and less.

USE PLATFORMIO on VSCODE

closed 2020-09-22 by ellensp

sjasonsmith, 2020-09-22:
One more thing you can do is change the temporary folder that Arduino builds in. Those paths make up most of the link length.

In your preferences.txt file, add something like this:

`build.path=C:\A\B`

To find your preferences file, open the Preferences dialog in Arduino, and you will see it at the bottom of the window.

`C:\A` is the folder I have Arduino in. Marlin is in `C:\M`. As @ellensp, the shorter the better.

You will probably still have to delete some Marlin source. Unused HALs and the `lcd/tft` folder are good candidates.