MarlinFirmware/Marlin #19467: [BUG] CR-10 MAX unable to home X when above Z+350mm

eXCommander, 2020-09-21:
CR-10 MAX CAN´T home when above Z+350mm!
X axis will keep banging the left frame brace unable to reach the switch! This should have been found and identified as a problem when designing the printer firmware. Every CNC engineer knows, that Z axis is to be referenced first!
So PLEASE CHANGE THE FIRMWARE, that this cant happen!
It damages the belt, pulley and can burn the motor. If someone operates the printer remotely, hes screwed. You have to physically reach for the switch to press it as even turning off the printer will not solve the issue as first thing it does after power on is homing and guess what? Its gonna keep banging the brace again! Z has to be referenced fist! Simple CNC RULE! On top of that it should always go Z+ to reference, but the design of printers is what it is so what ever...

thisiskeithb, 2020-09-21:
You didn’t follow the issue template which asked for your configs, but we don’t have a config for the CR-10 Max in the [Configurations repo](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality), so it’s not something we control.

You’ll need to contact Creality to request that they adjust their fork/config for that printer.

closed 2020-09-21 by thisiskeithb

InsanityAutomation, 2020-09-21:
You can specify the order of axis homing in the configuration file to an extent. That said, the machine is equipped with a probe and the head can move that tip off the bed. Also the bed leveling algorithm needs the home point to be set at the same point every time for proper referencing so Z_SAFE_HOMING is typically enabled to force the Z to require XY be homed and home Z to a known reference location for the bed leveling mesh. Homing Z when X/Y is in a bad place is even more dangerous than the attempted homing of X. If the tip is off the bed, it may go down ramming the head or nozzle into the build plate and damage the surface or the hotend.

It sounds like there is a mechanical Catch-22 in the design of the machine unfortunately.

eXCommander, 2020-09-21:
I dont have any files sorry... this is a stock coded printer. I did contact creality, but people also suggested to contact You. Thanks.

eXCommander, 2020-09-21:
Well the first thing is that the head should never be able to leave the constraints of the table platform. the bed should be fixed to the level of the printing head... which will be when im done with it...

thisiskeithb, 2020-09-21:
> Well the first thing is that the head should never be able to leave the constraints of the table platform

That's rarely true in 3D printing. It's pretty common for the nozzle/hotend to home off the build plate & Marlin has `X_MIN_POS`/`X_MAX_POS` and `Y_MIN_POS`/`Y_MAX_POS` to account for such hardware configs.

eXCommander, 2020-09-21:
I have the printer for less then 2 weeks, already working with octroprint etc doing the direct drive mod etc... Now it looks like i will have to do research on marlin as well... :D