MarlinFirmware/Marlin #19470: SD card not recognized as mass storage after m22

Johetan, 2020-09-21:
Hej, 
Just tried the newest Bugfix Version of Marlin, for linear advance and stuff (also latest build) 
Board is MKS SGEN-L

Since the firmware flash, while ejecting the sd with either m22 or on lcd the sd is not detected ass mass storage. 

Tried fiddling with the settings in configuration_adv.
Tried with onboard defined. Also with default for the board. 

Sd is recognized fine within Marlin, but not on the PC

thisiskeithb, 2020-09-21:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

Johetan, 2020-09-21:
1. Flashed the latest release of Marlin. 
2. Used the config example from the configuration github page
3. Changed some settings to suit my i2c oled 
In configuration.h I have sdsupport enabled
In configuration_adv I first tried with 
//#define SDCARD_CONNECTION ONBOARD
Then #define SDCARD_CONNECTION ONBOARD

The expected behaviour was, that I could read my sd card from my PC once the media is released. 
That is not the case.

thisiskeithb, 2020-09-21:
Please attach your configs.

Johetan, 2020-09-21:
> Please attach your configs.

Sorry, was on my Phone, just attached them :)

Johetan, 2020-09-22:
Found my Mishapp in Configuration.h 
Turning off the second Serial Port was the Solution. 
Thanks

closed 2020-09-22 by Johetan