MarlinFirmware/Marlin #19473: G35 Outputs long string then hangs firmware

RealDeuce, 2020-09-22:
### Bug Description

The tramming_point_name[] array is PROGMEM and points to PGMSTR() strings... this apparently means it can't be used with SERIAL_ECHOPAIR() on Arduino.

On my CR-10 v3 with original Creality board, the first output sent a long string of characters to the serial port and then hung.

### My Configurations

[MyConfig.zip](https://github.com/MarlinFirmware/Marlin/files/5259042/MyConfig.zip)

### Steps to Reproduce

1. Build for original Creality board with ASSISTED_TRAMMING
2. Execute G35 command over serial port.

**Expected behavior:** [What you expect to happen]

Instructions for tramming are printed

**Actual behavior:** [What actually happens]

The first tramming point name is printed as a very long sequence of letters followed by the instructions after which the firmware hangs and a reboot is required.

#### Additional Information

Fixed for me with [this commit](https://github.com/RealDeuce/Marlin/commit/486437ed5f6bf1644fc5802a1caa2a73a4b735ac), but the pull request requirements are too onerous for me to be arsed.  I know the patch isn't sufficient as-is, but I'm too busy to follow it through properly.

Basically, I just use serialprintPGM((char *)pgm_read_ptr(&tramming_point_name[i])); for the names and it works for me now.

sjasonsmith, 2020-10-26:
@Speaka reported the same issue in #19896 and verified that the included fix in this Bug report worked.
They also volunteered to prepare a PR for it, so hopefully we will see that show up in the next several days.

Speaka, 2020-10-27:
Please test with this PR.

sjasonsmith, 2020-10-28:
Thanks for stepping up and filing the PR @Speaka!

RealDeuce, 2020-10-28:
Works for me, thanks a lot @Speaka!

closed 2020-10-28 by RealDeuce