MarlinFirmware/Marlin #19477: [BUG] LCD becomes unresponsive during printing on BTT Mini SKR E3 v1.2

XDA-Bam, 2020-09-22:
### Bug Description

The LCD becomes unresponsive during printing. It beeps when pressing the encoder, but nothing happens for minutes. No animations, no screen updates. After 2 or  minutes, it may update and show the menu. It may react to scrolling once, but then immediately stops updating again.

### My Configurations

I am running an Ender 3 Pro with the board swapped for a BTT Mini SKR E3 v1.2. Stock display, firmware bugfix-2.0.x from today (47aa61c).

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5261442/Configs.zip)

### Steps to Reproduce

1. Install current bugfix
2. Reset EEPROM
3. Preheat for 10 minutes
4. Start a print from SD
5. Screen stops reacting as soon as the print is started.

**Expected behavior:** The screen should update (show animations and print progress) and react to inputs immediately.

**Actual behavior:** Screen basically freezes. It may update once every 3 minutes or so.

### Additional Information

Looks similar to #19186 at first glance. Different hardware, though.

swilkens, 2020-09-22:
which environment did you use in platformio.ini? I've seen some similar issues with the *_USB envs.

thisiskeithb, 2020-09-22:
> when running over serial.

There were several serial patches recently, so that should be resolved in the latest `bugfix-2.0.x`.

XDA-Bam, 2020-09-22:
I'm using `default_envs = STM32F103RC_btt`.

[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/5263840/platformio.zip)

boelle, 2020-09-22:
> There were several serial patches recently, so that should be resolved in the latest `bugfix-2.0.x`.

he is using a copy from today so it must still be there

sjasonsmith, 2020-09-23:
The serial fixes recently were hard hangs that would never recover. This issue sounds more like an overloaded CPU, not one that is completely locked up.

XDA-Bam, 2020-09-23:
> 
> The serial fixes recently were hard hangs that would never recover. This issue sounds more like an overloaded CPU, not one that is completely locked up.

Definitely not fully locked up. The display recovers as soon as the print is finished and it also occasionally updates during printing - every couple of minutes. Taking into account the surface artifacts which likely stem from slowdowns, this would point to CPU overload - I agree.

sjasonsmith, 2020-09-23:
Was this working with an earlier firmware? If so, can you identify a particular commit  or date when the problem began?

thisiskeithb, 2020-09-23:
Potentially related: I reverted some large buffer sizes in https://github.com/MarlinFirmware/Configurations/pull/168 due to a couple user complaints (I couldn't reproduce the issue on my end), so increasing them again may help.

XDA-Bam, 2020-09-24:
> 
> 
> Potentially related: I reverted some large buffer sizes in [MarlinFirmware/Configurations#168](https://github.com/MarlinFirmware/Configurations/pull/168) due to a couple user complaints (I couldn't reproduce the issue on my end), so increasing them again may help.

Yep. That 100% worked. I reverted all three buffers to the old values before @799f398 and as far as I can see, everything works flawlessly now.

Want me to check, which buffer is the problem?

thisiskeithb, 2020-09-24:
> Want me to check, which buffer is the problem?

Sure!

qwewer0, 2020-09-24:
What would be the benefit to have larger buffer values, or the disadvantage to have smaller buffer values? Would each have a different advantage/disadvantage?

XDA-Bam, 2020-09-24:
OK, I first reverted `BUFSIZE` to `4`. All good. Then `TX_BUFFER_SIZE` back to `0`. All good. Then `BLOCK_BUFFER_SIZE` back to `16` and the display froze pretty much immediately after the print began. So the problem is caused either by `BLOCK_BUFFER_SIZE = 16` alone, or the combination of those three lower buffer values.

I must say I'm wondering: I remember printing with this config on older firmware in july and it was mostly working fine. I think the display was already a bit laggy during printing and occasionally froze for a second or two. But now, the same config effectively freezes the display during printing.

brew99, 2020-10-01:
FWIW, I'm running bugfix from Aug 21 on a SKR mini E3 V1 board with BUFSIZE = 4, TX_BUFFER_SIZE = 0, BLOCK_BUFFER_SIZE = 16 and have not noticed any issues with the LCD display hanging.   I'm also using the default env of 512K with USB.   I've been running various bugfixs over the last year and haven't noticed any lcd hanging.   Printing is from SD card

XDA-Bam, 2020-10-01:
@brew99 Could you attach your config?

I mean, the board is different (v1 vs. v1.2), so maybe that's why you don't see this glitch. Or it could be the config. Or my board is somehow out of spec.

brew99, 2020-10-01:
@XDA-Bam 

This is what I am currently running on the V1.0 board without any issues.  It is obviously slower response on the LCD when printing, but nothing "hanging" like you describe. 

[configuration_adv.h](https://pastebin.com/raw/GV0SKK0A)

[configuration.h](https://pastebin.com/raw/CLR79U6g)

qwewer0, 2020-10-01:
I didn't experience this issue with default or changed buffer values, but maybe my configs can help.

Ender 3, SKR Mini E3 v1.2
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5314662/Configuration.zip)

XDA-Bam, 2020-10-01:
OK, so the configs look as follows:

Setting | XDA-Bam | brew99 | qwewer0
-|-|-|-
Environ | 256K | 512K_USB | 512K
Buffer | 32/32/32 | 16/4/0 | 16/4/0
JD | Off | On | On
S-Curve | Off | On | Off
LA | On | On (K=0) | On
Microstep | 32/32/16 | 16/16/16 | 16/16/16

My money is on the microstepping. I'm running X & Y with 32, which could be problematic in conjunction with a lower `BLOCK_BUFFER_SIZE` of `16`. Could technically be linked to Jerk/JD, because this is also different between our configs, but jerk should be easier on the processor than JD.

I will see if I can check this on the weekend. If you're bored, try running 32 microsteps with the default buffer size for a print to check it out (don't forget to double your steps/mm).

brew99, 2020-10-01:
@XDA-Bam 

I tried 1st altering the XY microstepping to yours 32/32/16, with my normal buffers 16/4/0 and it was fine, no issues or change.

2nd I then added the buffers of 32/32/32, keeping microstepping at 32/32/16 (which is what you are running and see the LCD hanging) and the result is the same.  I don't have any hanging of the LCD on either of these configs.   Ran each one for about 1 min or print time and dove into the LCD menu system and it all acted normal.

XDA-Bam, 2020-10-04:
OK, so I updated to bugfix-2.0.x @ 604d10d3 today and tested it again. Results are inconclusive:
Block_Buffer	| X&Y Microsteps	| Result
-|-|-
32		| 32			| Freeze during purge line, recovers
16		| 32			| Freeze during purge line, sometimes freezes during print
16		| 16			| Freeze during purge line, recovers

Freeze in this context means, that the display doesn't respond anymore or only delayed (>10 s). Either some recent commits improved the outcome, or the freezes are more random than I thought.

I'm pretty much good with my current config and `BLOCK_BUFFER_SIZE=32`. It appears to make the menu more responsive regardless of the initial freezes. If no one has any particular interest in this problem, I'd close this up.

closed 2020-10-11 by XDA-Bam