MarlinFirmware/Marlin #19478: [BUG] Backlash Compensation Skipping Steps

stabbs, 2020-09-22:
I just updated my printer's firmware to version 2.0.6.1 to take advantage of backlash compensation on the z axis.  I have a worm / belt driven z bed and there is 1.94mm of backlash measured with a dial indicator.  I am also using x,y, and z extruder offset for tool changes.

  #define BACKLASH_DISTANCE_MM { 0, 0, 1.94 } // (mm)
  #define BACKLASH_CORRECTION    1.0       // 0.0 = no correction; 1.0 = full correction

The z stepper motor sounds like its trying to spin faster depending on the reversing distance.  I posted a YouTube link below showing this issue as well.  For example:
G1 Z1.0 F100
G1 Z0.0 F100    -  Backlash compensation works well, no issues

G1 Z1.0 F100
G1 Z0.9 F100   -  Motors strain and skip some steps

G1 Z1.0 F100
G1 Z0.99        -  Motors strain and skips all steps

I have my z speed limited to 5mm/s and acceleration at 120mm/s but the backlash compensation ignores these limits.  I have also tried enabling BACKLASH_SMOOTHING_MM 3 but I need the backlash compensation to be immediate for successful tool changes.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5262160/Configuration.zip)

Please see this video demonstrating the backlash problem
https://youtu.be/MUfKBlZGgww

Thanks in advance for the help and apologies if this is posted in the wrong place

boelle, 2020-09-22:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

stabbs, 2020-09-22:
Thank you for your reply.  I'm getting the following error:  'Error compiling for board Arduino Mega or Mega 2560'
Followed[ these instructions](https://forum.arduino.cc/index.php?topic=621875.0) here to clear the compile error but the error persists

`Here's the fix:
1.	Tools > Board > Boards Manager
2.	Wait for downloads to finish.
3.	From the list of boards platforms, click "Arduino AVR Boards".
4.	Click the "Update" button.
5.	Wait for installation to finish.
6.	Click "Close".`

sjasonsmith, 2020-09-22:
That happens due to command length limitations on Windows that Arduino doesn’t handle well. There are ways to work around it, but it is better to learn how to build Marlin using Visual Studio code and PlatformIO.

stabbs, 2020-09-22:
**bugfix-2.0.x branch** did not fix or change the backlash compensation issue posted above

Sidenote: I couldn't get **bugfix-2.0.x branch** to compile in Arduino IDE 1.9.0 beta but was able to compile and upload using  PlatformIO.  Also compile times were 16 minutes in Arduino IDE and only 3 in Platform IO

sjasonsmith, 2020-09-22:
If you disable backlash compensation do those small moves work properly?

From the video it definitely sounds like acceleration is not being applied correctly for small moves, I just want to very whether backlash compensation is actually an important factor in reproducing this.

stabbs, 2020-09-23:
I think this issue is related to the steps/mm.  My current setup uses a worm gear with a 30:1 reduction so my z-steps is set to 2445.  I reduced this to 1000 and the backlash compensation seems to work fine- no skipping or whirring.  It starts buzzing/skipping around 1800 steps

closed 2020-11-01 by github-actions[bot]