MarlinFirmware/Marlin #19479: [BUG] Z tower does not do a second bump while homing Delta marlin 2.0.6.1

motymen, 2020-09-22:
### Bug Description

when homing X and Y towers will do a slow second homing , Z tower will oinly do a fast home 

### My Configurations

arduino 
ramps 1.6 plus (runing as 1.4) 
reprap discount graphical LCD 
drv8825 drivers 
marlin 2.0.6.1
config: [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5262442/Marlin.zip)

### Steps to Reproduce

1. unpack files to marlin 2.0.6.1
2. flash into ramps
3. perform home from pc or LCD menu

**Expected behavior:** all 3 towers do a fast home and then a slow homing

**Actual behavior:** Tower X and Y perform a fast then a slow homing, Tower Z Performs only a fast home

#### Additional Information

video attached: [homing video.zip](https://github.com/MarlinFirmware/Marlin/files/5262490/homing.video.zip)

please disregard the graphics problem, (there is a problem with LCD...) error occours when lcd is disconnected (and homing from pronterface)

boelle, 2020-09-22:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

Also use vscode+plaform.io as that is the prefered tool

motymen, 2020-09-29:
im new quite new with github so sorry if answers are abit off

about the bug:
the second i uncommented  "#define DELTA_AUTO_CALIBRATION" the problem disappeared now its homing as it should

about the buxfix branch:
ive checked "bugfix-2.0.x branch" there is no mention of a homing bug for delta so i don't know what else to do there

about the vscode+plaform.io:
is it a must? yes the arduino tool sucks a bit(lot) but unless it could be the cause for the bug im not changing it

zonalimitatore, 2020-09-29:
No,because im using platformio,last marlin version,and,i have no way to do the second probe (but do in the ubl process)
ive no delta,but the problem its the same

sjasonsmith, 2020-10-28:
This would be easy to try to reproduce. I am going to assign my name to it so I can find it again and try it this weekend, if nobody has solved it before then.

zonalimitatore, 2020-10-28:
Isn't a bug. I've "play" in marlin's config with the distance between probing,z height after,ecc...and now work

sjasonsmith, 2020-10-28:
@zonalimitatore thanks for trying to help, but in this case I think your comparisons are only applicable if homing a delta. I’m not willing to state there is no issue without anybody else even attempting it on a delta.

sjasonsmith, 2020-10-31:
I failed to reproduce this with both 2.0.6.1 and bugfix-2.0.x (a4a487b0cc0a431c01ac07c8170b8a5f19cffd2e ).

In both cases I tested both with and without `DELTA_AUTO_CALIBRATION`. Our printers are different enough that that I couldn't match your configurations exactly, but I didn't see any significant differences that should relate to homing.

At this point I think we need you to try to reproduce with your printer on the `bugfix-2.0.x` branch. I know there are no commits which would seem to have impacted this behavior, but if a fix is needed that is going to be done on top of the bugfix branch.

sjasonsmith, 2020-11-07:
@motymen we haven't heard anything from you in over a month. Based on testing on my own printer, I will assume you have resolved the issue. Please reply if this is not true and you can still reproduce problems with the `bugfix-2.0.x` branch.

closed 2020-11-07 by sjasonsmith