MarlinFirmware/Marlin #19481: [BUG] Z_CLERANCE_DEPLOY_PROBE + Z_HOMING_HEIGHT = Z_OFFSET adjust impossible

wzaburko, 2020-09-22:
After 2 weeks  maybe I found problem with my z-offset :)
I had problems after homing Z and G0 Z0 nozzle was 3.5 mm above bed. But my Z offset is 3.5mm:).  If I adjusted my z offset to correct position Z offset was next -3.5mm (total Z=-7mm). After next homing Z, nozzle was one more time 3.5mm above bed. Never ending story.
In my config was:
USE_PROBE_FOR_Z_HOMING
Z_CLERANCE_DEPLOY_PROBE = 10
Z_HOMING_HEIGHT = 10

But I tried set Z_CLERANCE_DEPLOY_PROBE = 0 because I use  BLtouch to homing  and I think not need 20 mm gap.
And problems gone on Marlin 2.x bugfix and 2.0.61 :)

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5263172/Config.zip)

### Steps to Reproduce

1. G28 Z
2. move nozzle to bltouch position
3. G0 Z0
4. Adjust Z_offset form LCD panel
5. G28 Z
6. move nozzle to bltouch position
7. G0 Z0

**Expected behavior:**
Nozlle touch bed
**Actual behavior:** 
Nozzle is over bed 

#### Additional Information

Board: FLYF407ZG
TMC2209 (MKS v.2.0)
BLTouch Smart v3.1

mikemerryguy, 2020-09-22:
for sure config issue...
Z_CLERANCE_DEPLOY_PROBE isn't even defined in 2.0.6.1

but #define Z_CLEARANCE_BETWEEN_PROBES 5 is default
Z_HOMING_HEIGHT = 4 is disabled by default 

Your issue is that your Nozzle is above bed, that is intended behavior

it probes the Z then PARKS the nozzle at a set HT

if you want you can 
//#define Z_AFTER_HOMING  10      // (mm) Height to move to after homing Z
to
#define Z_AFTER_HOMING  0      // (mm) Height to move to after homing Z

Not at all recommended.

You simply need to send a command to move Z to 0 then measure and adjust your Z offset from there.

wzaburko, 2020-09-23:
I do this in point 3 and 7.

closed 2020-11-01 by github-actions[bot]

Lukas1818, 2020-11-25:
I have this still this issues. My Marlin is on version 65983b42886e08e29f6281e243a0f33923487a93.
After home the z-axis will not be drive up. There is an offset of -10mm now and I can 
drive down the z-axis to -10mm (my value for Z_AFTER_HOMING) and hit my printbed.
May this issue should be reopen.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5599167/Configuration.zip)