MarlinFirmware/Marlin #19482: [BUG] Mesh edit with activate/deactivate UBL - nozzle is not in same position

wzaburko, 2020-09-22:
If I try to adjust mesh manually form LCD panel, if UBL is active nozzle is over bed (I think) by  Z-offset value. This is after G28 Z and turn off UBL or G28 turn on UBL.

### My Configurations
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5263336/Config.zip)

### Steps to Reproduce

1. G28 Z
2. G29 P1
3. form LCD panel - Motion/UBL/ turn form Activate UBL  (to deacitvate :) - is not better UBL On/Off?
4. form LCD panel - Motion/UBL/Edit Mesh
5. set and observe the positions of the selected spot
6. G 28 Z
7. form LCD panel - Motion/UBL/ turn form Deactivate UBL  (to acitvate :)
8. form LCD panel - Motion/UBL/Edit Mesh
9.  set and observe the positions of the selected spot

**Expected behavior:** 
turn off UBL before Mesh Edit
**Actual behavior:** 
If is UBL on nozzle above bed if is UBL off is on correct position

zonalimitatore, 2020-09-29:
Before do a g29,i think you need to be sure that ubl was off

wzaburko, 2020-09-30:
I did, but I forgot about it a few times and wasted a lot of time.  Now i add g28a in gcode without activating ubl on marlin startup and is ok.

Roxy-3D, 2020-10-31:
> I did, but I forgot about it a few times and wasted a lot of time. Now i add g29a in gcode without activating ubl on marlin startup and is ok.

It would seem to me that the LCD Mesh Edit should do the 'Right' thing whether UBL is activated or not.    The fact of the matter is the Mesh Edit is going to temporarily disable (and probably restore) UBL activation in order to do the edit.

Can you check and verify that it behaves differently in each of the two modes?   And if so...   I'll get it patched so it behaves deterministically.

closed 2020-12-26 by github-actions[bot]