MarlinFirmware/Marlin #19483: "Home Z" in Marlin 2.0.x

roweder, 2020-09-22:
I apologize if this is not the right way to request help with this. I did not think reporting this as a bug was correct, but I also believe it's a feature that already exists, so this might not be right either.  

I've read a decent bit on the internet already about adding a menu option for homing only Z, or only X & Y, but all the posts I've found predate Marlin 2.0.x, and my Marlin 2.0.x configuration.h, configuration_adv.h, and ultralcd.cpp files don't have the same homing menu code as I've seen in lines 2706 to 2710 in the [ultralcd.cpp file for marlin 1.1.x](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/ultralcd.cpp).  I also can't find where the "Auto Home" menu item is defined in Marlin 2.0.x configuration.h, configuration_adv.h, and ultralcd.cpp files.  

How can I add options to home individual axes in 2.0.x, and to which files should I add the code?

Any assistance is greatly appreciated,

-Derek

qwewer0, 2020-09-22:
Would `INDIVIDUAL_AXIS_HOMING_MENU` do? Configuration.h line 1810.

roweder, 2020-09-22:
> Would `INDIVIDUAL_AXIS_HOMING_MENU` do? Configuration.h line 1810.

Wow that was fast, thank you!  That should do it!

qwewer0, 2020-09-22:
That will give you individual homing option for X, Y and Z. If you only want one or don't want some, then you can comment out what you don't need in menu_motion.cpp line 329.

If this solves your problem, then you can close this issue.

roweder, 2020-09-22:
Thank you so much qwewer0, the language in menu_motion.cpp looks much more like what I've seen in posts that pre-dated Marlin 2.0.x.  It is now fully clear to me how to do this, thanks again for your help!  I'll close the issue!

closed 2020-09-22 by roweder