MarlinFirmware/Marlin #19487: [BUG] Wrong NOZZLE_TO_PROBE_OFFSET value stored in EEPROM M851

IvSan, 2020-09-23:
### Bug Description

Wrong NOZZLE_TO_PROBE_OFFSET value stored in EEPROM M851.
My probe is on the left and front from nozzle, so I'm setting
`#define NOZZLE_TO_PROBE_OFFSET { -40, -12, 0 }`
After compiling and updating the printer I'm reading EEPROM with `M503`. Result:
`M851 X40.00 Y12.00 Z0.00`
And auto bed leveling do not work correctly until I update the values with `M851 X-40.00 Y-12.00 Z0.00`

### My Configurations

https://github.com/IvSan/Marlin_config

### Steps to Reproduce

1. Set some negative values into `NOZZLE_TO_PROBE_OFFSET`
2. Compile and update the printer firmware
3. Read it with `M503`, you will receive `M851` with positive values

**Expected behavior:** `M851` with negative values

**Actual behavior:** `M851` with positive values

#### Additional Information

Ender 3 SKR Mini E3 V2.0
Latest bugfix branch (same on 2.0.6.1)

ellensp, 2020-09-23:
EEPROM values take precedence over firmware values. If you update NOZZLE_TO_PROBE_OFFSET in Configuration.h you NEED to refresh the contents of your EEPROM. Send the machine a M502 then a M500. 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-23 by ellensp

IvSan, 2020-09-23:
Sorry, but I do not understand how is storing absolute value but loosing negative sign can be a normal behavior.

IvSan, 2020-09-23:
#19437 looks related

IvSan, 2020-09-23:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/394 May be related

sjasonsmith, 2020-09-23:
@ellensp is assuming that your eeprom has a stale value in it. You have to re-initialize the EEPROM to use a new value from the header.

IvSan, 2020-09-23:
Realize that M502 will bring settings from configuration, not mysterious factory settings.
Thanks folks. Sorry for being dumb.

sjasonsmith, 2020-09-23:
Nothing dumb about it. It is obviously a usability issue because stale eeprom settings gets people all the time. I even fall for it once in a while, even though I know better!