MarlinFirmware/Marlin #19491: [BUG] [2.0.6] "Stop Print" sucks back filament

FlailAway, 2020-09-23:
### Bug Description
After a couple of contiguous "Stop Print" and restart, it takes about 40+ seconds for filament to appear at the nozzle.

### My Configurations
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce
Not sure I can do that exactly, yet, but this is what's happening so far. 
It seem to be worse after 2 or 3 successive Stops.
Run a print from the SD card.
Stop the print at layer 1 or 2.
Head pauses then lifts, X retracts, Y retracts spewing filament all the way. Extruder still feeding.
Pop out SD, make a change to gcode and put it back.
Purge the filament manually until visible at the hotend.
Start the print. It now goes for about 40+seconds and no filament as it has been sucked back quite a distance.
Latest time I had around 50-seconds per first layer and the filament did not appear until half way through the second layer.
Only happens after a Stop or 2.

**Expected behavior:** [What you expect to happen]
Leave the dang filament alone. When the Stop is issued, just stop the filament before doing anything else.
Don't retract before or on the next start.

[stop.zip](https://github.com/MarlinFirmware/Marlin/files/5270372/stop.zip)

GMagician, 2020-09-23:
It seems like you miss some "extruder" reset position in your gcoce. is this possible? This may be the reason why rebooting it works. Once rebooted E pos is 0

FlailAway, 2020-09-23:
> It seems like you miss some "extruder" reset position in your gcoce. is this possible?

Thanks, but IdeaMaker Slicer has that set up in their boilerplate, so always included.

`
;TYPE:SKIRT
;WIDTH:0.400
G1 F2400 E0.0000

`

ellensp, 2020-09-23:
exactly.. you purge and purge and E is at some length say 15. you then tell it to got back to E0 and it retracts!  
you need to tell it it is already at E0 with a G92 E0

closed 2020-09-23 by ellensp