MarlinFirmware/Marlin #19495: [FR] BTT SKR 1.3 EXP-MOD config

phentrophil100, 2020-09-24:
Need modify pins_BTT_SKR_*.h for support  EXP-MOT v1.0 
[Pin-diagram](https://ae01.alicdn.com/kf/H401273477f0140c790059f4e824b36f7o.jpg)
I tried to do it according to the pattern from connecting the display, but I don't know how correct this is :
`/**
 * EXP-MOT CONFIG UART
 *               _____                                              _____
 *           NC | 1 2 | GND                                     NC | 1 2 | GND
 *           NC | 3 4 | M1EN                                  M2EN | 3 4 | M3EN
 *        M1STP | 5 6   M1DIR                                 M1RX | 5 6   M1DIAG
 *        M2DIR | 7 8 | M2STP                                 M2RX | 7 8 | M2DIAG
 *        M3DIR | 9 10| M3STP                                 M3RX | 9 10| M3DIAG
 *               -----                                              -----
 *               EXP2                                               EXP1
 */

#define EXP1_3_M2EN                       P1_23
#define EXP1_4_M3EN                       P1_22
#define EXP1_5_M1RX                       P1_21
#define EXP1_6_M1DIAG                       P1_20
#define EXP1_7_M2RX                       P1_19
#define EXP1_8_M2DIAG                       P1_18
#define EXP1_9_M3RX                       P0_28
#define EXP1_10_M3DIAG                       P1_30

#define EXPA2_3_PIN                       -1
#define EXP2_4_M1EN                       P1_31
#define EXP2_5_M1STP                       P0_18
#define EXP2_6_M1DIR                       P3_25
#define EXP2_7_M2DIR                       P0_16
#define EXP2_8_M2STP                       P3_26
#define EXP2_9_M3DIR                       P0_15
#define EXP2_10_M3STP                       P0_17`

ellensp, 2020-09-24:
firstly https://github.com/bigtreetech/BTT-Expansion-module/tree/master/BTT%20EXP-MOT

ellensp, 2020-09-24:
I don't think this should be added to pins...  all those pins are already defined. 
more add it as a module
with config parameters like 
#define HAS_BTT_EXP_MOT
#If(HAS_BTT_EXP_MOT)
#define BTT_EXP_MOT_A DRIVER_TYPE  
#define BTT_EXP_MOT_B DRIVER_TYPE  
#define BTT_EXP_MOT_C DRIVER_TYPE
#define SET_BTT_EXP_MOT_A_AS X2
#define SET_BTT_EXP_MOT_A_AS Y2
#define SET_BTT_EXP_MOT_A_AS Z2
#endif 
and the module sets the appropriate axis pins to the already defined exp pins.

thisiskeithb, 2020-09-24:
> and the module sets the appropriate axis pins to the already defined exp pins.

Making it smart so it'd check if users still have an LCD defined that uses those pins/throw an error would probably also be a good idea.

sjasonsmith, 2020-09-25:
> Making it smart so it'd check if users still have an LCD defined that uses those pins/throw an error would probably also be a good idea.

I imagine you would make it an `else` case and if you enabled this expansion module you would not actually define any of the LCD pins.

phentrophil100, 2020-09-25:
> > Making it smart so it'd check if users still have an LCD defined that uses those pins/throw an error would probably also be a good idea.
> 
> I imagine you would make it an `else` case and if you enabled this expansion module you would not actually define any of the LCD pins.

agree with this.

And I would like to know what pins I can bind to tx cs for each of m1,m2,m3?

ellensp, 2020-11-10:
Can you give the above PR a try.
Just add #define HAS_BTT_EXP_MOT 1  to your Configuration.h

ellensp, 2020-11-11:
Is now part of marlin.

closed 2020-11-11 by ellensp

kyaron, 2020-12-02:
I enabled #define HAS_BTT_EXP_MOT 1, but the stepper motor connected to the EXP board doesnt turn. 
crossed cables, motors, driver none helped.
Any idea???