MarlinFirmware/Marlin #19496: Help! Big Tree Tech SKR Mini E3 V2.0  + Reprapdiscount smart controller = White Squires

NickRosato, 2020-09-24:
Hello,

I have recently been working on a custom 3d printer based on some old parts that I had from previous builds. I recently got a Big Tree Tech SKR Mini E3 V2.0 and was trying to adapt it to my Reprapdiscount smart controller (small white one). After powering it on I only get white boxes and no startup text. 

Things I have done
-Use Deflate configure for SKR Mini E3 V2.0 
-Undefined define CR10_STOCKDISPLAY
-define REPRAP_DISCOUNT_SMART_CONTROLLER
-Added #elif ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER)  in Marlin-2.0.x-SKR-mini-E3-V2.0\Marlin\src\pins\stm32f1\pins_BTT_SKR_MINI_E3_common.h
-Set pins in both EXP1 and EXP2 (did not plug sd in) based on wiring from https://reprap.org/wiki/RepRapDiscount_Smart_Controller
-Tested with volt meter and on LCD screen pins based on 
   -Pin 1 5v: VDD
   -Pin 2 Ground: K
   -Pin 3 (LCD_EN) PB15: E
   -Pin 4 PB8  (LCD_RS): RS
   -Pin 5 (LCD_D4) PB9 : D4
-Powered up -> only noticed squires as seen in picture
-Clicked POT = Beep
-Clicked Reset/stop = Board Restart

Any help regarding what I have to do in order to get this LCD to work would be appreciated. I feel like Marlin is not telling the display to do anything, but I am unclear how to change that in my Config.

[image]

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/5273785/Configuration.txt)
[pins_BTT_SKR_MINI_E3_common.txt](https://github.com/MarlinFirmware/Marlin/files/5273787/pins_BTT_SKR_MINI_E3_common.txt)

ellensp, 2020-09-24:
Not enough pins defined...

lcd alone is 
#define LCD_PINS_RS 16 //[RAMPS14-SMART-ADAPTER]  
#define LCD_PINS_ENABLE 17 //[RAMPS14-SMART-ADAPTER]  
#define LCD_PINS_D4 23 //[RAMPS14-SMART-ADAPTER]  
#define LCD_PINS_D5 25 //[RAMPS14-SMART-ADAPTER]  
#define LCD_PINS_D6 27 //[RAMPS14-SMART-ADAPTER]  
#define LCD_PINS_D7 29 //[RAMPS14-SMART-ADAPTER]

sjasonsmith, 2020-09-24:
It seems like creating custom cables and pin definitions for displays and boards which are physically incompatible are beyond the scope of a Marlin bug report.

ellensp, 2020-09-24:
That lcd need more pins thannyou have available
6 for lcd
3 for encoder 
1 for beeper
1 for kill switch
5 for sdcard.
+ pwn and gnd

NickRosato, 2020-09-24:
That is what I figured, I just purchased a big tree display that is comparable. Thank you for reassuring my thoughts.

closed 2020-09-24 by NickRosato