MarlinFirmware/Marlin #19497: [BUG] M33 for certain files names return ???

chilman408, 2020-09-24:
### Bug Description

When I issue the M33 command on certain files to get the long file name, I'm getting a ??? in return.  This is specifically for files that I have sliced on my Windows 10 PC.  Other gcode files with long names that I downloaded off the internet would return the correct file name with M33.  I can't figure out what is going on.  I'm not using any special characters in my file-names, just plain old English.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-09-24:
Please attach your Marlin configuration files, as requested by the issue template. Marlin behavior is extremely configurable, and it is very difficult to debug any issue without knowing how the firmware is configured.

Try enabling `SCROLL_LONG_FILENAMES`.

chilman408, 2020-09-24:
Attached.  `SCROLL_LONG_FILENAMES` is enabled.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5274422/config.zip)

ellensp, 2020-09-24:
can you zip up a example file that doesn't list   (you can delete all gcode in the file, shouldn't make any difference ) So it effectively just a filename.

chilman408, 2020-09-24:
Yes, thx.

Attached (included gcode).
[sample-gcodes.zip](https://github.com/MarlinFirmware/Marlin/files/5277108/sample-gcodes.zip)
T2TFT35V2P0330.gcode (slicded on my computer) would return ???
k-factor_calibration.gcode which was download off the net would return M33 fine.

rhapsodyv, 2020-09-25:
I saw a similar issue when working with the asset loader for LVGL UI. What I found is that some files are saved in the SD without the attribute `DIR_ATT_LONG_NAME`:
```cpp
Marlin/src/sd/SdFatStructs.h:579
static inline uint8_t DIR_IS_LONG_NAME(const dir_t* dir) {
  return (dir->attributes & DIR_ATT_LONG_NAME_MASK) == DIR_ATT_LONG_NAME;
}

... checked here:
Marlin/src/sd/SdBaseFile.cpp:1099
 // Fill the long filename if we have a long filename entry.
    // Long filename entries are stored before the short filename.
    if (longFilename && DIR_IS_LONG_NAME(dir)) {
```

I didn't go deep looking to see if the attr is really missing or if is marlin doing a bad reading. 

In my case, all files fit in the 8.3 scheme, so I just did a fallback to the dosFileName when longFilename[0] == 0:
```
Marlin/src/lcd/extui/lib/mks_ui/pic_manager.cpp:494
        // If we dont get a long name, but gets a short one, try it
        if (card.longFilename[0] == 0 && d.name[0] != 0)
          dosName2LongName((const char*)d.name, card.longFilename);
```

We call apply the same fallback to the M33.

I know it don't fix the problem, but I think it is better than `???`.

The origin of the problem could be the SD format too.

@chilman408 Did you try with different SD cards? Maybe re-formatting it with FAT 16 or FAT 32...

rhapsodyv, 2020-09-25:
Found!! hahah.
the problem is the T2 in the name of the file... !!! it is interpreted as a command because M33 was not in the list of commands that was able to receive a unescape string... 

Confirmed and fixed: #19515

chilman408, 2020-09-25:
I see thank you

PR #19515 (2020-09-26): Fix M33 string without escape
### Description

M33 wasn't in the list of commands that were able to receive a unescaped string... so, some filenames were interpreted as commands.... 

### Benefits

Fix #19497

### Related Issues

#19497
closed 2020-09-25 by chilman408