MarlinFirmware/Marlin #19499: [FR] (feature request title)

qwewer0, 2020-09-24:
### Description

Would it be possible to make M350 - Set micro-stepping to work with TMC drivers or to make a dedicated TMC command to set micro-stepping? (e.g. SKR Mini E3 v1.2, TMC 2209)

thisiskeithb, 2020-09-24:
No. `M350` is for drivers that have digital microstepping pins which the SKR mini boards do not have. https://marlinfw.org/docs/gcode/M350.html

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

sjasonsmith, 2020-09-24:
@thisiskeithb why close this? Seems perfectly valid as a feature request. We had already discussed it on Discord.

thisiskeithb, 2020-09-24:
> @thisiskeithb why close this? Seems perfectly valid as a feature request. We had already discussed it on Discord.

This seemed like a question about M350's use and not a feature request.

Since you're aware/handling it, I'll reopen it.

qwewer0, 2020-09-24:
Changed the description to be more like a FR and not a question.

thisiskeithb, 2020-09-24:
Following the [Feature Request template](https://raw.githubusercontent.com/MarlinFirmware/Marlin/2.0.x/.github/ISSUE_TEMPLATE/feature_request.md) really helps 🙂

sjasonsmith, 2020-09-25:
The M350 command on RepRap seems to be documented to allow up to 256 microsteps. It doesn't clarify whether it works with TMC drivers, but to support 256 microsteps that seems incredibly likely.
https://www.reprap.org/wiki/G-code#M350:_Set_microstepping_mode

Yep, looks like they are using M350 to set both microstepping and interpolation on TMC drivers:
https://github.com/Duet3D/RepRapFirmware/blob/bcb169fbf6fc8457afb272cd8ddf3d6b76d58795/src/Movement/StepperDrivers/TMC22xx.cpp#L568

I imagine to be useful it would also have to store settings to EEPROM.

qwewer0, 2020-09-25:
@sjasonsmith So, that means it works for TMC drivers already, and can close this FR, or while it works, it still should be able to make it work on boards that has TMC drivers without digital micro-stepping pins?

sjasonsmith, 2020-09-25:
I was just looking at what RepRap firmware does with the command. I wasn’t referring to Marlin code at all. There is some desire to  use the same commands across different firmwares where possible. RepRap configuring this through M350 makes it easier to justify adding it to Marlin.

I think M350 would be the place to do it, the remaining question is whether making it runtime configurable is a good idea. Every constant which becomes configurable makes the math a little harder while printing, which may be noticeable on slower controllers.

teemuatlut, 2020-10-03:
I don't think it would be used during printing. But I question if this command is something that would be used more than just during initial setup and I think this is just not something that is "tuned". It's basically a set it and forget it type of configuration item.
Then if you want to have the command for setting this live, then you'll likely also want it to be saved in EEPROM and overall I just don't see the point.

qwewer0, 2020-10-03:
One use case I can think of right now, is a "cold pull" gcode. If the extruder isn't strong enough on 16 micros-tepping to pull the filament out, then switching it to full stepping would result stronger extrusion, and thus would be able to pull the filament out. Assuming this is how the stepper motor torque changes relation to the micro-stepping.

teemuatlut, 2020-10-03:
Microstepping doesn't decrease torque.
While you're taking finer steps, you're also taking comparatively more steps and the end result is the same delta between rotor and stator, thus resulting in the same applied torque.

qwewer0, 2020-10-03:
So you are saying that micro-stepping doesn't decreases the rotation torque, only the holding torque? And so it would pull the filament with the exact same amount of force?

mihtjel, 2021-04-03:
Probably gonna look into implementing this tomorrow or Sunday as I want to try out some different settings and can't be bothered recompiling and flashing just for that.

(Yes, yes. I am going to implement this feature _because I'm lazy_, despite it doubtless being much more work.)

mikhail11111, 2022-02-15:
Is there any progress here?

ellensp, 2022-02-15:
You do realize with most tmc's they are already set to   #define INTERPOLATE      true
So the only effect that changing the micro stepping does is to change the loading on the micro controller?

DiskDoctor, 2022-05-02:
Is M350 enabled for >16 yet? Looking to find a way to identify TMC drivers via software vs tearing apart printers and heat sinks. 
If M350 S256 (for example) 
can be used, TMC's should respond, A4988's should report error. Is this possible?

ellensp, 2022-05-02:
@DiskDoctor take a look at macro AXIS_IS_TMC(A)

DiskDoctor, 2022-05-02:
@ellensp  Thanks. Could you kindly point me to the macros? I've used search but no results. Discord, another link?

thisiskeithb, 2022-05-02:
> I've used search but no results.

Run a global search in VSCode:
[image]

mikhail11111, 2022-05-03:
> You do realize with most tmc's they are already set to #define INTERPOLATE true So the only effect that changing the micro stepping does is to change the loading on the micro controller?

Right, hence the maximum speed attainable