MarlinFirmware/Marlin #19502: [BUG] Wrong number of extruders if MIXING_EXTRUDER is selected

radek8, 2020-09-24:
If the option is enabled
#define MIXING_EXTRUDER
#if ENABLED (MIXING_EXTRUDER)
#define MIXING_STEPPERS 2 // Number of steppers in your mixing extruder
The printer reports the M115 command
EXTRUDER_COUNT: 1
If I use the touch screen via the serial port, the printer does not detect the second extruder. I do not have the opportunity to control the second extruder.
Isn't this a mistake?
The problem is solved here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1055

thisiskeithb, 2020-09-24:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

SerjPr, 2020-09-25:
> Please follow the Issue Template when filing a bug report. This includes attaching your configs:
> 
> ```lisp
> 
> 
> ### Description
> 
> 
> 
> ### Steps to Reproduce
> 
> 
> 
> 1. [First Step]
> 2. [Second Step]
> 3. [and so on...]
> 
> **Expected behavior:** [What you expect to happen]
> 
> **Actual behavior:** [What actually happens]
> 
> #### Additional Information
> 
> * Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
> * Provide pictures or links to videos that clearly demonstrate the issue.
> * See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
> ```
-->

### Description

### Steps to Reproduce

<The MIXING_EXTRUDER option is enabled. The system has two physical extruders and one mixing head, one heating element and one temperature sensor.
The monitor sends a request to the M115 marlin to determine the number of extruders in the system. But gets the response EXTRUDER_COUNT: 1>

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** If the option is enabled
#define MIXING_EXTRUDER
#if ENABLED (MIXING_EXTRUDER)
#define MIXING_STEPPERS 2 // Number of stepper motors in your mixing extruder

Printer reports command M115
EXTRUDER_COUNT: X
Where X is the number of extruders specified in #define MIXING_STEPPERS

**Actual behavior:** EXTRUDER_COUNT:1

#### Additional Information

#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
#define MIXING_STEPPERS 2 // Number of steppers in your mixing extruder
#define MIXING_VIRTUAL_TOOLS 16 // Use the Virtual Tool method with M163 and M164
//#define DIRECT_MIXING_IN_G1 // Allow ABCDHI mix factors in G1 movement commands
#define GRADIENT_MIX // Support for gradient mixing with M166 and LCD
#if ENABLED(GRADIENT_MIX)
#define GRADIENT_VTOOL // Add M166 T to use a V-tool index as a Gradient alias
#endif
#endif
Enabling these settings is enough for the mixing extruder to work. If I enable #define EXTRUDERS 2 in the settings, it will lead to an error.

boelle, 2020-09-25:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

> The problem is here:
> [bigtreetech/BIGTREETECH-TouchScreenFirmware#1055](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1055)

So to be clear, is the problem with marlin or BTT ?

Vertabreak, 2020-09-25:
this is no error or bug mixing is done virtually with vtools with 1 virtual extruder that is defined by the active vtool this is working as intended and this report should be considered user error.

closed 2020-09-25 by ellensp

SerjPr, 2020-09-25:
> Конфигурации, пожалуйста ,
> пожалуйста , застегивать свои `Configuration.h`и `Configuration_adv.h`файлы (как это
> предусмотрено в шаблоне выпуска) и поместите их в свой следующий ответ.
> Мы их проверим и посмотрим, что не так.
> 
> Пожалуйста, проверьте, [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)где он стоит.
> 
> > Проблема тут: 
> > [bigtreetech / BIGTREETECH-TouchScreen Прошивка # 1055](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1055)
> 
> Итак, чтобы прояснить, проблема в марлине или BTT?

BTT determine the number of extruder motors in the received response "EXTRUDER_COUNT:1" on request "M115
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5283807/Configuration_adv.zip)

". And only after that it turns on the ability to control the motor of the second extruder. By what demand can BTT determine how many extruder motors are in the system?