MarlinFirmware/Marlin #19503: HardwareSerial.h causing UART communication problems in SKR1.3/4 (again)

Drzet, 2020-09-24:
Last update to platformio introduced changes to HardwareSerial.h for the LPC176x boards.
Previously UART pins were directly (wrongly) defined whereas now `define LPC_PINCFG_UART3` must be used.
Problem is, it is unclear **where this definition must be placed?**
This change has caused my SKR1.4 to stop communicating with WiFi ESP board as default pin assignments are not the pins in the wifi slot.

The only (not so graceful) way to solve this that worked for me was to comment all the "if's" and alternative definitions out.
See the code below

` } else if (UARTx == LPC_UART3) {
      // Initialize UART3 pin connect
      PinCfg.OpenDrain = 0;
      PinCfg.Pinmode = 0;

      //#if defined(**LPC_PINCFG_UART3_P4_28**)
        PinCfg.Funcnum = 3;
        PinCfg.Portnum = 4;
        PinCfg.Pinnum = 28;
        PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 29;
        PINSEL_ConfigPin(&PinCfg);
      //#elif defined(LPC_PINCFG_UART3_P0_25)
        //PinCfg.Funcnum = 3;
        //PinCfg.Portnum = 0;
        //PinCfg.Pinnum = 25;
        //PINSEL_ConfigPin(&PinCfg);
       //PinCfg.Pinnum = 26;
        //PINSEL_ConfigPin(&PinCfg);
      //#else
       // PinCfg.Funcnum = 2;
       // PinCfg.Portnum = 0;
        //PinCfg.Pinnum = 0;
        //PINSEL_ConfigPin(&PinCfg);
        //PinCfg.Pinnum = 1;
        //PINSEL_ConfigPin(&PinCfg);
      //#endif`

thisiskeithb, 2020-09-24:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

daleckystepan, 2020-09-24:
Just set
#define SERIAL_PORT 3
#define SERIAL_PORT_2 -1

And you dont need any special settings. ESP in Wifi slot on SKR 1.4 will work out of the box.

p3p, 2020-09-24:
@Drzet I'm not understanding what your issue is, If you are building with Marlin `bugfix-2.0.x` then you should't have to do anything for the correct UART pins to be selected by the framework, we have added the ability to specify this (framework build flags) is the boards pins file. 

If you are not using `bugfix-2.0.x` (or a version new enough) then you will need to add `-DLPC_PINCFG_UART3_P4_28` to the `build_flags` option in the `[common_LPC]` `platformio.ini` environment to utilise this new feature in the lpc arduino framework.

Drzet, 2020-09-24:
> @Drzet I'm not understanding what your issue is, If you are building with Marlin `bugfix-2.0.x` then you should't have to do anything for the correct UART pins to be selected by the framework, we [...]

My issue is: how on earth am I supposed to find that out ? The only information provided in HardwareSerial is "define LPC_PINCFG_UART3" comment. 
Thank you for the explanation.

p3p, 2020-09-24:
>  how on earth am I supposed to find that out 

When something isn't supposed to be changed by normal users I'm afraid the comments (if there are any) can become a bit undescriptive, and that goes does double when the option is part of the Arduino framework and not Marlin.

Do you know if there are other boards other than the SKR 1.4 that use non standard pins for UART?

Drzet, 2020-09-24:
SKR1.4 turbo and 1.3 I guess

Drzet, 2020-09-30:
> @Drzet I'm not understanding what your issue is, If you are building with Marlin `bugfix-2.0.x` then you should't have to do anything for the correct UART pins to be selected by the framework, we [...]

I have now recompiled my firmware with `bugfix-2.0.x` branch . I have noticed #define BOARD_CUSTOM_BUILD_FLAGS -DLPC_PINCFG_UART3_P4_28 in the .pins file.
Unfortunately ESP3D doesn't work again. Do I need to add the build_flag manually?

thisiskeithb, 2020-09-30:
Everyone reporting issues: Please follow the issue template above and attach your configs.

By not providing them, you are making it difficult for anyone to help you.

daleckystepan, 2020-09-30:
> > And you dont need any special settings. ESP in Wifi slot on SKR 1.4 will work out of the box.
> 
> Please attach your configs.

https://github.com/daleckystepan/Marlin/commits/geeetech_a10_32bit

Drzet, 2020-09-30:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5307581/configs.zip)

Drzet, 2020-10-01:
Tried my previous solution, to comment out all alternative configurations in HardwareSerial - does not work.
Tried with another ESP board - no luck either

closed 2020-10-17 by Drzet