MarlinFirmware/Marlin #19504: [BUG] Example import-2.0.x/config/examples/delta/Tevo Little Monster wrong board defined

lemmingDev, 2020-09-24:
### Bug Description

The the example configuration for Tevo Little Monster, the MOTHERBOARD is defined as BOARD_MKS_GEN_L

The motherboard that comes with all Tevo Little Monsters is the MKS SBase, so the definition should be BOARD_MKS_SBASE

### My Configurations

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5279144/Configs.zip)

### Steps to Reproduce

1. Use the provided Tevo Little Monster Configuration.h and Configuration_adv.h files

**Expected behavior:** [What you expect to happen]
#define MOTHERBOARD BOARD_MKS_GEN_L

**Actual behavior:** [What actually happens]
#define MOTHERBOARD BOARD_MKS_SBASE

#### Additional Information
There could be other changes needed as well,

thisiskeithb, 2020-09-24:
Looks like they've been broken since day one: https://github.com/MarlinFirmware/Marlin/pull/12370

What else in the config should be updated once you change `MOTHERBOARD` to `BOARD_MKS_SBASE`?

lemmingDev, 2020-09-24:
Hi

That was the first thing I could find. After changing that, It compiled in PlatformIO and successfully installed on the board using the firmware.bin on microsd, and it added the FIRMWARE.CUR file, which is the indication that it uploaded.

After that however, I was unable to connect to the board using the supplied MKS TFT touchscreen.

I'll take another stab at it in the next few days after I've printed the stuff I need (as Smoothieware is currently working fine - but would like Marlin) and report back

sjasonsmith, 2020-09-25:
I was surprised that we would have the wrong board, especially since it is in the direction of a downgrade to an 8-bit board.

This Tevo site seems to agree that they come with an SGEN:
https://help.tevo.cn/downloads/firmware/

lemmingDev, 2020-09-25:
From the site above you listed, it indeed says that the TEVO Little Monster comes with the SBASE (Not SGEN)

[image]

I guess it was someone wanting to run Marlin 1.X on the printer, and that only supported 8bit boards...

sjasonsmith, 2020-09-25:
You are right. I read one name and typed the other.

sjasonsmith, 2020-10-25:
@lemmingDev I attempted to update this for you, but I am not comfortable with the changes I have to make, without having the printer to test it.

Do you have a working configuration for a stock Little Monster?

The details that I am unsure about:
- Microsteps? 32 or 64? (And the steps/mm derived from this)
- Axis and endstop inversion?
- BLTouch configuration

lemmingDev, 2020-10-26:
https://raw.githubusercontent.com/Homers3D/Tevo-Little-Monster/master/Control%20Board/config

I'm pretty sure all the info can be gleaned from there

Haven't had a chance to have a crack at it myself yet...

boelle, 2020-11-22:
@lemmingDev had a chance to have a crack at it?

lemmingDev, 2020-11-22:
I've got 6 weeks off coming up in 3 weeks.

Will take a look at it then...

rhapsodyv, 2020-12-02:
Is it a marlin bug or a configuration error in the examples repository? Show we move this to the Configurations repo?

thisiskeithb, 2020-12-02:
It's a configuration issue, but there may need to be two configs since Tevo updates the boards in their printers over time.

For example: The Tevo Tornado used to ship with an MKS Base, then MKS Gen-L, and now they are shipping with an MKS SGen-L.