MarlinFirmware/Marlin #19506: [BUG] (Ender 5 Pro Config in Configurations-bugfix-2.0.x - multiple issues )

seanwhiteau, 2020-09-25:
### Bug Description

The latest Marlin sample configuration has incorrectly inverted the steppers in the sample configuration for the Ender 5 Pro. 
In "Configuration.h":

**#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR true**

should be instead:
**#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR false**

Also **#define INVERT_E0_DIR true**
should be instead:
**#define INVERT_E0_DIR false**
So currently, the extruder is working in reverse.

This IF statement is missing from the CR10_STOCKDISPLAY area:

#if ENABLED(CR10_STOCKDISPLAY)
  #define RET6_12864_LCD
#endif

Very sloppy hacking!

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-09-25:
It sounds like you have the new version with a V4.x board.

The current config is for the older version with the 128k melzi board and `CR10_STOCKDISPLAY` (not RET6 LCD).

Creality caught everyone offguard with these new 4.x boards & releasing the same printer models with these in them, so configs for every possible variant don't exist yet.

seanwhiteau, 2020-09-25:
At the very least, it needs one minute worth of code comments.

thisiskeithb, 2020-09-25:
> At the very least, it needs one minute worth of code comments.

We'll need to create additional configs for the new boards/printers instead of marking up the existing ones.

seanwhiteau, 2020-09-25:
Creality are sending the 4.2.7 and 4.2.2 upgrade boards randomly now... so you don't know what you're going to get until it arrives. Their Marlin firmware on their website is a nightmare too. Thanks Keith!

cosmicfilament, 2020-09-25:
Hmm
I have the 4.2.2 silent board on an ender 5 pro and my configuration.h settings are different than the above
This is the first time I have built the firmware, but it has been running great for the last week.

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_CREALITY_V4 //modified
#endif

#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false //modified

#define INVERT_E0_DIR false //modified

#define X_HOME_DIR -1 //modified
#define Y_HOME_DIR -1 //modified
#define Z_HOME_DIR -1

#define USE_XMIN_PLUG //modified
#define USE_YMIN_PLUG //modified
#define USE_ZMIN_PLUG //modified
//#define USE_XMAX_PLUG //modified
//#define USE_YMAX_PLUG //modified
//#define USE_ZMAX_PLUG //modified

#define CR10_STOCKDISPLAY
#define RET6_12864_LCD //modified

john

thisiskeithb, 2020-09-25:
Creality ships the Ender-5 with Ender-3 settings, but the Ender-5 homes to X and Y max, not min. Those defaults put the X0,Y0 origin in the right-rear, which is not correct and throws off logic like the probe offsets, margins, etc.

This is a commonly debated, but `X_HOME_DIR`/`Y_HOME_DIR` should be 1, not -1 and matching `USE_XMAX_PLUG`/`USE_YMAX_PLUG`. Also, `USE_Z_MIN_PLUG` should be used since the printer homes to Z min (distance between nozzle and bed decreases as you home) even if it looks like it homes to Z max due to how the bed rises.

We’ve corrected these key settings in the Marlin Configurations repo so the X0,Y0 origin is now correctly located in the front-left corner.

cosmicfilament, 2020-09-25:
Hey thanks for the info cited above.
I don't really have much depth to my knowledge as I am new to 3d printing.
Which led me to think that back right was correct as that was how the firmware that was shipped was configured.

I just made it work.

John

mikemerryguy, 2020-09-26:
this is awesome stuff...
creality was lazy with firmware for all ender 5 variants and just made the end stops 00 at back right!!!
 
they also didn't account for E steps with upgraded metal extruder and left it at default 93 as the original stock ender 3 had plastic extruder mount!!

they also don't assemble hot ends properly to eliminate gaps!!!

overall decent, but not AT ALL assemble and print ready machines!!!!

ellensp, 2020-09-26:
@seanwhiteau you never mentioned what motherboard you actuality have. Eithera  V4.2.2 or a V4.2.7
If it is a V4.2.7  I have just created a PR to add this machine with this motherboard to the config examples. And a few other Ender's too.  https://github.com/MarlinFirmware/Configurations/pulls

seanwhiteau, 2020-09-26:
@ellensp - I received a 4.2.2 upgrade board although I actually ordered a 4.2.7. 

I managed to get mine working with my clumsy hacks, but these builds will certainly help other people, thank you. The Marlin firmware that Creality are offering on their website for these new boards doesn't work properly for a lot of people. In my case, the Creality compiled Marlin firmware made a constant beeping sound, and my add-on BLTouch wouldn't work at all. Creality didn't bother including the source with these firmware builds, which I just assumed was them just being disorganised.
Thank you, Sean

cosmicfilament, 2020-09-26:
I ordered a bltouch right after I bought my ender 5 pro. I sent it back after futzing with it for a day. Not sure how a newbie could figure out how to make it work. And forget about the host of conflicting bin files on their website(s).
Now I use manual mesh bed leveling and it works really great. I think that I could get the bltouch working now if I re-ordered it. But no need.

seanwhiteau, 2020-09-27:
@cosmicfilament - if you can get a BLtouch going, I think it is much better than doing it manually. This video might help: https://www.youtube.com/watch?v=neS7lB7fCww&list=PLkajNWoAdkKaWZyUiebrPj73Pt7txkJr_&index=1&t=1155s
If you have a 4.2.7, then you could use ellensp's configurations (above) as a starting point, and use the YouTube video as a guide on how to tweak it further for the BLTouch. You really have to make sure you have the wires right. The comments under that YouTube video will help you too.

cosmicfilament, 2020-09-27:
@seanwhiteau - I have watched that video. Actually, I have watched lots of his videos. They are pretty good. Thanks for the feedback. Maybe I will give it another go.

ArsenicBismuth, 2020-10-16:
@cosmicfilament Wow, I should've get into this thread rather than researching myself lol. Got into the exact same conclusion, except my invert is not working?!! No matter if I set it to True or False, moving back-right are positives.

So I'm forced to use origin at MAX, rather than the default (0,0). Anyone know why? It's funny how inverting parameter just won't work.

thisiskeithb, 2020-10-16:
Closing since there is now a 4.2.7 config for the Ender-5: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-5/CrealityV427

And to reiterate my [comment above](https://github.com/MarlinFirmware/Marlin/issues/19506#issuecomment-699187786) about the Ender-5 settings being wrong out of the box:

> Creality ships the Ender-5 with Ender-3 settings, but the Ender-5 homes to X and Y max, not min. Those defaults put the X0,Y0 origin in the right-rear, which is not correct and throws off logic like [...]

closed 2020-10-16 by thisiskeithb

ellensp, 2020-10-16:
From what I've been able to determine its identical to 4.2.7 config just with a different motherboard line.
ie #define MOTHERBOARD BOARD_CREALITY_V427 becomes #define MOTHERBOARD BOARD_CREALITY_V4 for the V4.2.2

kalvinpearce, 2020-10-16:
@ellensp You are a life saver. This really saved my weekend. Just tested a print and looked like that one line does the trick and I finally have manual mesh bed leveling (unfortunately after much pain failing to get it working yesterday I already bought a BLTouch 😂) Although such a minor change to get it working, I would love to see the V427 config become a V422/V427 config with a define to select which board you have as it would save many hours of heart break for some. If there is no issues with this/you also think its a good idea I am happy to make a PR with the relevant changes and comments

seanwhiteau, 2020-10-16:
In the new sample config for 4.2.7 at https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-5/CrealityV427  I noticed that:
**#define SHOW_CUSTOM_BOOTSCREEN**
in _Configuration.h_ should be commented out.

ellensp, 2020-10-16:
@seanwhiteau  
no it shouldn't be commented out. It has a custom boot screen file _Bootscreen.h

seanwhiteau, 2020-10-16:
@ellensp  - my mistake, apologies. I hadn't changed the _Bootscreen.h correctly