MarlinFirmware/Marlin #19508: Laser PWM is constant regardless of velocity

mirozmrzli, 2020-09-25:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5280944/config.zip)
based off commit b3223d7b412b0067e0f9c1ad7b672317098a973f (origin/bugfix-2.0.x, bugfix-2.0.x)

board: skr 1.4 turbo LPC1769 TMC2209 drivers

#define SPINDLE_LASER_PWM_PIN   P1_00
#define LASER_FEATURE
#define LASER_POWER_INLINE
#define LASER_POWER_INLINE_TRAPEZOID_CONT
#define LASER_MOVE_POWER

Expecting the laser PWM to follow the current motion speed, starting at 0% duty cycle at 0 speed and being what's set by S parameter in G1 command at max speed.

Observed PWM is constant regardless of speed.

test gcode:

    M201 X20
    G1 X000 F10000 S127
    G1 X100 F10000 S127
    G1 X000 F10000 S127
    G1 X100 F10000 S127

Resulting waveform looks like this regardless of whether stepper just started moving or it's at full speed:
[image]

mirozmrzli, 2020-09-28:
Ok, turns out that both

    LASER_POWER_INLINE_TRAPEZOID
    LASER_POWER_INLINE_TRAPEZOID_CONT
and 

    LASER_POWER_INLINE_TRAPEZOID_CONT_PER 0
need to be defined for PWM to start changing. However, M5 doesn't turn the laser pwm off:

    G0 X0 Y0
    M201 X15 Y15
    M3 S127 I
    G1 F1000
    G1 X50
    G1 Y50
    G1 X0
    G1 Y0
    M5

this leaves PWM at about 6% duty cycle most of the time. Same if M5 is omitted. If those commands are typed in manually through terminal (using pronterface), it works fine. Only way to reliably turn laser off is to use

    LASER_MOVE_G0_OFF

and issue dummy G0 move

shitcreek, 2020-09-29:
With `LASER_MOVE_POWER` you can just use 'G1' with 'S' param - you don't need the 'M3'

Have you tried passing 'I' with your 'M5' since you're passing 'I' with 'M3'?

mirozmrzli, 2020-09-29:
> 
> 
> With `LASER_MOVE_POWER` you can just use 'G1' with 'S' param - you don't need the 'M3'
> 
> Have you tried passing 'I' with your 'M5' since you're passing 'I' with 'M3'?

Just tried, using 'M5 I' with 'M3 I' also leaves laser on, just as using S with G1 and M5

closed 2020-12-03 by github-actions[bot]