MarlinFirmware/Marlin #19512: [BUG] Can not build firmware 2.0.6 for Ender 3 V4.2.7

Follll, 2020-09-25:
Hallo, I try build firmware 2.0.6 for Ender 3 V1.5 with board V4.2.7 and have some issues

[image]
[image]
[image]

Can you help me please?

Thaks

boelle, 2020-09-25:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. 

But there are a clue in the first pic, read the red text

ellensp, 2020-09-25:
You clearly have not updated the configuration.h file with a new #define MOTHERBOARD  BOARD_CREALITY_V427
It looks like the only thing you changed is the build environment in platfomio. 
Perhaps you copied the new config files to the wrong directory?

ellensp, 2020-09-25:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-25 by ellensp

Follll, 2020-09-25:
> Komu: MarlinFirmware/Marlin <Marlin@noreply.github.com>
> Ender 3 V4.2.7 (#19512)
> "

Hallo,
thak for your reaction. configuration files are here. I try bugfix-2.0.x 
branch(https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x), but it 
is the same error.. I use configuration files from included cofig files for 
buffix marlim 2.0.X

My 3D printer is Ender 3 PRO and now I have board Creality V 1.1.5 with 
marlin 2.0. and i want to change board to 4.2.7. but Marlin 2.0.6.1 is not 
function..

Thak you

Petr
---------- Původní e-mail ----------

(https://github.com/MarlinFirmware/Marlin/issues/19512#issuecomment-698986819)
"

thinkyhead, 2020-09-25:
@Follll — This is the most appropriate reply to your question:

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).