MarlinFirmware/Marlin #19518: Ender 5 z axis won't go up after installing 427 board and bltouch

emartin8907, 2020-09-26:
can't get it to auto home, the z will not go up when I enable the following in marlin bugfix 2.0

Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
bltouch
auto_bed_beveling_bilinear
z_safe_housing
babystep_zpro

Using this guide https://www.youtube.com/watch?v=neS7lB7fCww&feature=youtu.be

When I disable 'z_min"probe_uses"z_min_endstop_pin' it will go up, but then crashes into the bed/ the bltouch pin doesn't deploy...

boelle, 2020-09-26:
Did you read the template? (try create a new issue but not post it)

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

emartin8907, 2020-09-26:
Hi boelle thanks for chiming in, I just uploaded my config files to my main post, sorry I'm new here.

I am not sure what you mean by test the bugfix 2.0 branch?  I am using Marlin bugfix 2.0

boelle, 2020-09-26:
Hi, bugfix 2.0 is just as i could not see what you are using

but dinner is close here so i would suggest you do an extra check on the wires and that you test the endstops etc

emartin8907, 2020-09-26:
wiring all looks fine, i tried swapping the white and black wires on the bltouch but that didnt change anything.  

Not sure how to test endstops?

boelle, 2020-09-26:
bl touch black goes to endstop gnd
white goes to endstop signal

M119 gives status of endstop

touch one with finger and send M119 again

do that for all endstop to make sure they work and report correctly

boelle, 2020-09-26:
from LCD you can tell it to deploy and stow bl touch, if it does that it works correctly

if it do not work after this there must be a config error 

i have a core xy with bl touch and it works fine

emartin8907, 2020-09-26:
i can't tell which is the ground or signal on the board.

now it wont deploy on startup or when I tell it to on the lcd screen.  going to start over in marlin...

emartin8907, 2020-09-26:
okay well I've re done it twice now and the z goes up but just crashes.  and my bltouch wont deploy at all not on startup or anything

Am I supposed to update these bits of code even using bugfix 2.0 ? https://github.com/InsanityAutomation/Marlin/commit/15ce74badfd3a1b6e6ffabf882234ffa77682715
I thought I would try it but I couldn't even find marlin_main.cpp does that still exist in 2.0?

boelle, 2020-09-26:
maybe @InsanityAutomation could chime in when he have time

siana, 2020-09-26:
You can use M119 to check endstop states. But it won't tell you much because BLTouch may correctly report triggered when stowed. However it being triggered and not deploying is the reason the homing move completes instantly.

The right way around for the wiring should be when STOPPED error, not when the bed crashes into the nozzle. And then figure out why it isn't deploying. You can use M280 to try various servo positions and see whether you can get it to deploy.

It may be relevant which version BLTouch is, and whether it's even genuine.

There is **M43 S** to check BLTouch function.

ellensp, 2020-09-27:
The CrealityV4.2.7 comes with TMC2208_STANDALONE stepper drivers.  you have not updated your config files stepper drivers to this.

ellensp, 2020-09-27:
Also since these are standalone you need to adjust the vref manually with the variable resistor. I have seen it before that the vref was set just to the edge of moving, where with gravity it would move down, but against gravity it wouldn't move up.

emartin8907, 2020-09-27:
thanks guys.  I've rebuilt a few times today from scratch, and still can't get the bltouch to come down at all.  It is the 3.1 version, with blue/ red wire that needed to be swapped.  It came in a kit on amazon.  just to be sure I just ordered a new one from actual ANTCLABS on amazon, should be here tomorrow, looks like it has the brown/red wires already in the right place.  

I changed my driver_types to TMC2208_STANDALONE, thanks
Right now it will go up but, since the bltouch isn't working it would just crash if I didn't stop it.  Hoping new one tomorrow just magically fixes it.  
I haven't seen anyone mention adjusting the vref manually with this setup (board/enders) I will have to look into this I don't know anything about that, but than you!

ellensp, 2020-09-27:
From what I can tell this is all support, not bugs..  as such I direct your attention to the following. 
I recommend the discord to talk to people real time 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-27 by ellensp