MarlinFirmware/Marlin #19521: [BUG] parameter [E<bool>] at G29 does not work as described

lonelymyp, 2020-09-26:
I am using a pressure sensor for testing. 
Before each point, I need to zero the sensor readings because the error is accumulating.
For zeroing, I use the solenoid enable signal.

### Bug Description
The solenoid signal turns on at the star of probing, then the axis travels to the coordinates of the first probe point.

### My Configurations
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5287033/Configuration_adv.zip)

### Steps to Reproduce
G29 E1

**Expected behavior:** 
Movement of the axles with the solenoid off signal
Turn on the solenoid signal only after the axis stops, exactly before deploy.

**Actual behavior:** 
The carriage moves from the current position to the first point with the signal on.

marlin 2.0.6.1
mks gen_l 2.1

lonelymyp, 2020-10-27:
up

boelle, 2020-11-22:
> marlin 2.0.6.1

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sjasonsmith, 2020-12-24:
I'm going to keep this one alive because I have a printer with a solenoid probe that needs this to work properly to avoid exceeding the solenoid's duty cycle. I'll have to try it out next time I am with that printer.

@lonelymyp you will probably be interested in the newly added `PROBE_TARE` feature, which seems to address your use-case perfectly. It will be a pulse rather than being active for the duration of the probe, but I suspect that might work even better for what you need.

closed 2021-02-01 by github-actions[bot]