MarlinFirmware/Marlin #19524: [BUG] Adapting working config from 2.0.5.3 to 2.0.6.1 crashes Marlin when issuing G28 Z0 command

chilman408, 2020-09-27:
### Bug Description

Unsure what is changed between 2.0.5.3 and 2.0.6.1 but when my configs are adapted to the new version and FW flashed to machine, when I perform an auto home, when command G28 Z0 is issued.  Marlin crashes and requires a restart.

### My Configurations

See attached.
[2.0.5.3.zip](https://github.com/MarlinFirmware/Marlin/files/5287623/2.0.5.3.zip)
[2.0.6.1.zip](https://github.com/MarlinFirmware/Marlin/files/5287624/2.0.6.1.zip)

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-09-27:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2020-09-27:
After updating, are you Initializing EEPROM either through the LCD or by sending `M502` followed by `M500`?

Vertabreak, 2020-09-27:
i would also note G28 Z0 would be considered a invalid command the correct command would be G28 Z the addition of the 0 makes the command invalid.

https://marlinfw.org/docs/gcode/G028.html

thisiskeithb, 2020-09-27:
> i would also note G28 Z0 would be considered a invalid command 

True, but Marlin shouldn’t crash. At least it doesn’t crash on my test machines when issuing `G28 Z0`.

chilman408, 2020-09-27:
Thanks everyone.

@ellensp - Compiled the bugfix version.  Still crashes Marlin but went a bit further than before because the BLTouch needle extended before crashes.

@thisiskeithb - yes, EEPROM is re-initialized after a flash.

@Vertabreak - Thanks, I didn't know that but this is the command that Octoprint issues when I try to home the Z axis.  I used Auto Home on the LCD as well and same crash happens when try to Z-home.

Any other ideas?  It is so weird that going from 2.0.5.4 (I compiled this version last night as well; going from 2.0.5.3) to 2.0.6.1 would cause this to happen.

So to re-cap, Marlin worked fine up to 2.0.5.4.

thisiskeithb, 2020-09-27:
Here’s the next level of troubleshooting:

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

chilman408, 2020-09-27:
Ok, thx, will report back.

chilman408, 2020-09-28:
> Here’s the next level of troubleshooting:
>
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
>
> * Download Marlin [...]

Hello,
Attached are the log output from a couple of runs using G28, G28  X Y and G28 Z....
[Trial 2.txt](https://github.com/MarlinFirmware/Marlin/files/5295583/Trial.2.txt)
[Trial 3.txt](https://github.com/MarlinFirmware/Marlin/files/5295584/Trial.3.txt)
[Trial-1.txt](https://github.com/MarlinFirmware/Marlin/files/5295585/Trial-1.txt)

sjasonsmith, 2020-10-25:
I think I just reproduced this in the experimental simulator. Here are the configs I used, which were adapted from the user's posted configuration.

To reproduce, I simply booted the simulator and issues `G28 X Y`.

[Simulator_Reproduce.zip](https://github.com/MarlinFirmware/Marlin/files/5434452/Simulator_Reproduce.zip)

If the simulated failure is actually the same root cause, something related to S_CURVE_ACCELERATION is causing division by zero in stepper.h. Disabling S_CURVE_ACCELERATION prevents the error.

[image]

thinkyhead, 2020-10-25:
Interesting! I'll work on getting the latest simulator working with the debugger on my system later today and mess around with it. I'm already finding the sim super useful for quickly developing and testing graphical LCD stuff.

sjasonsmith, 2020-10-25:
@thinkyhead it looks like the failure I encountered was a bug in the simulation code. The DELAY_CYCLES mechanism allowed simulated interrupts to run recursively. Once I patched that up homing is fine.

@chilman408 a lot has changed in the past four weeks, could you test with the latest bugfix again?

I'm trying to understand your configs. You specify the motherboard as `BOARD_SOVOL_SV01`, but that board isn't present in Marlin. Where is your source from?

chilman408, 2020-10-25:
wow, thank you guys.  Yes, I'm using the source from here:
https://github.com/CopterTec/Marlin-Sovol_SV01/releases

Although, I remember using the OEM source with the same result.

thinkyhead, 2020-10-25:
@chilman408 —

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sjasonsmith, 2020-10-28:
@chilman408 that fork has not been updated since July, and might be based on even older Marlin.

I am going to close this, since we cannot support forks of Marlin.

We can reopen this if you are able to reproduce the issue using the bugfix-2.0.x branch from this repo, and can provide all the files needed to reproduce it.

closed 2020-10-28 by sjasonsmith