MarlinFirmware/Marlin #19525: [BUG] Case Lights work with gcode, but not in menus

ZhuDaHai, 2020-09-27:
### Bug Description

I have 24v LED strip lights wired into HEV1 on an MKS SGEN L v2.0 32bit mainboard. 
And an BTT TFT35 v3.0 display running in 12864LCD emulation mode wired to EXP1 and EXP2
Display is configured in Marlin as REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and appears to be working properly.

I can control the LED strip lights (case lights) via gCode command:

`M106 P1 S255`

And can set various levels of brightness with S successfully.

I set configuration_adv.h as follow to allow control of the case light through the 12864LCD display:

```
#define CASE_LIGHT_ENABLE                     // DaHai - Enabled for LED Strips
#if ENABLED(CASE_LIGHT_ENABLE)
  #define CASE_LIGHT_PIN P2_06                // Override the default pin if needed // DaHai: Using HE1
  #define INVERT_CASE_LIGHT true              // Set true if Case Light is ON when pin is LOW // DaHai: was false
  #define CASE_LIGHT_DEFAULT_ON true          // Set default power-up state on
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 255   // Set default power-up brightness (0-255, requires PWM pin) // DaHai: was 105
  //#define CASE_LIGHT_MAX_PWM 128            // Limit pwm
  #define CASE_LIGHT_MENU                     // Add Case Light options to the LCD menu // DaHai: So I can set via the menus
  //#define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.
  //#define CASE_LIGHT_USE_NEOPIXEL           // Use Neopixel LED as case light, requires NEOPIXEL_LED.
  #if ENABLED(CASE_LIGHT_USE_NEOPIXEL)
    #define CASE_LIGHT_NEOPIXEL_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
  #endif
#endif
```
In pins_MKS_SGEN_L_V2.h, FAN1 is defined as follows:

```
if HOTENDS == 1
  #ifndef FAN1_PIN
    #define FAN1_PIN                       P2_06
  #endif

```

But when I try to control the lights through the 12864LCD menus, they flicker when I change the brightness of remain off.
They also remain off even when I toggle from OFF to ON in the menu.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5287778/Marlin.zip)

### Steps to Reproduce

Go to the 12864LCD menu to the Lights menu
set different intensities for the light
turn the lights on and off

**Expected behavior:** [What you expect to happen]
I expect the lights to turn on and stay on, not flicker off

**Actual behavior:** [What actually happens]
Lights flicker off and won't stay on

#### Additional Information

I have also tried setting  INVERT_CASE_LIGHT to false and true, in configuration_adv.h, but it had no effect.

I expect there is a conflict between controlling the lights via gcode M106 P1 and through the menus, but I don't know how to resolve, or if this is an actual bug.

Thank you for your time.

ellensp, 2020-09-27:
You cant have it setup as a fan and a case light...   the fan code will turn it off if fan gcode is not used.
Comment out #define FAN1_PIN                       P2_06

ellensp, 2020-09-27:
use https://marlinfw.org/docs/gcode/M355.html to control case light from gcode.

ZhuDaHai, 2020-09-27:
> You cant have the pin setup as a fan and a case light... the fan code will turn it off if fan gcode is not used.
> Comment out #define FAN1_PIN P2_06

Thank you!
Works perfectly now!
Figures I was causing the conflict 🙁

closed 2020-09-27 by ZhuDaHai