MarlinFirmware/Marlin #19527: [BUG] Ender 3 V2 - First flash from stock, unable to change LCD language to English.

CRCinAU, 2020-09-27:
### Bug Description

There have been numerous reports of new Ender 3 V2 printers being upgraded to Bugfix and are unable to change the LCD language from Chinese to English after the flash.

Flashing a version built on 2020-09-15 allows the screen language to be changed - and any further updates to newer bugfix nightlies will correctly operate when changing the language of the LCD.

This does not work on versions built on 2020-09-19 onwards.

### My Configurations

Stock as per:
https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20V2

### Steps to Reproduce
See video: https://youtu.be/Q5PT3BjYobU

Vertabreak, 2020-09-27:
Has the standard eeprom wipe been performed post flash as has always been required to clear out the old no longer valid eeprom settings and load the new. 
M502 + M500

CRCinAU, 2020-09-27:
The DWIN LCD language is not stored within the Marlin EEPROM structure - however, in Marlin, it is stored at the EEPROM address 0x01.

See:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/dwin/e3v2/dwin.cpp#L208

CRCinAU, 2020-09-27:
I was originally thinking that this line here was the fault:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/dwin/e3v2/dwin.cpp#L217

ie reading a `uint8_t` into a `bool` - however I have been told that anything non-zero when converted to a bool is considered true.

I haven't managed to find anything else that stands out - HOWEVER, once you managed to set the screen to English (ie Marlin writes a 0x00 or 0x01 to address 0x01), then this function works perfectly on newer builds.

As such, I thought it must have something to do with the contents of that EEPROM address BEFORE Marlin has written a valid value to it...

CRCinAU, 2020-09-28:
@thinkyhead - I believe this may be in your refactors between 15/09 and 19/09 commits - but given that we're looking at hundreds of lines, I haven't been able to single out the commit as yet...

Sono-Gomorrha, 2020-09-28:
I have the same issue and M502 + M500 has no effect. I also tried updating the LCD firmware (which should not play a role) and it did not change anything.

thinkyhead, 2020-09-28:
I've applied a patch that will set the Ender 3 V2 language to English by default, including previous installs with the language set to Chinese, so give the latest `bugfix-2.0.x` code a try.

CRCinAU, 2020-09-28:
Have had reports that this indeed throws the screen into English.

thisiskeithb, 2020-09-28:
Closing as issue is resolved.

closed 2020-09-28 by thisiskeithb