MarlinFirmware/Marlin #19530: [FR] Filament width sensor calibration

hamster65, 2020-09-27:
As far as I understand, Marlin expects the measured filament diameter as a voltage. So 1,75 V means 1,75 mm etc.
This requires the filament width sensor to have an microcontroller for calibration / scaling. You cannot just connect an analog sensor to the ADC.

There was also an issue with 3.3 V boards like LPC1786: https://github.com/MarlinFirmware/Marlin/issues/14680
I've got an  LPC1786 3.3 V board and with an open (unconnected) ADC input, it shows a fluctuating diameter of around 4.7 mm. Since the ADC only goes to 3.3 V, it should not measure more than 3.3 mm under any condition. So I assume it has not been fixed and the scaling is still wrong for 3.3 V ADCs/boards.

Both problems could be solved with an internal calibration feature.
The procedure would be like this (for 1.75 mm printers):
1) Insert a small reference object into the sensor, say a 1.5 mm drill shaft.
2) Send Gcode1 to the printer, with argument 15 for the width of the reference object.
3) Insert a larger reference object into the sensor, say a 2 mm drill shaft.
4) Send Gcode2 to the printer, with argument 20 for the width of the reference object.

Both inputs and the sensor value applied to a map function, should give the actual filament diameter. That should work for any ADC reference voltage / range and any analogue sensor, provided it has a linear output. Even if it doesn't, I think it should be close enough. A third measurement could also be added in the middle, let's say with 1,75 mm filament measured with a caliper.

Optionally, menu entries for calibration and sensor on/off.

thisiskeithb, 2020-09-27:
Link to original bug report: https://github.com/MarlinFirmware/Marlin/issues/14680

hamster65, 2020-10-12:
Would it make sense to combine it with a runout feature? When the diameter is too low -> runout.

hamster65, 2020-11-23:
I am trying to adapt the code so that it works with 3.3V boards. However, I have never looked into the depths of Marlin and I am not a programmer.

I connected a potentiometer between GND and 3.3V and the wiper to the TX pin auf the Aux connector of my MKS S GEN L.
#define FILWIDTH_PIN P0_02_A7 //AUX1 TX-Pin

In src/feature/filwidth.h, line 70 there is
static inline float raw_to_mm(const uint16_t v) { return v * 5.0f * RECIPROCAL(float(MAX_RAW_THERMISTOR_VALUE)); }

Changing 5.0f to 3.3f gives wrong results for the filament width. I thought that maybe it relies on the 10 bit ADCs of the AVR controllers, so I divided by 4. Still way off. Dividing by 16 gives the right reading. Could it be that this function assumes an 8 bit ADC with 5V reference voltage? In this case it should give wrong readings even on 5V 8 Bit AVR boards because they have a 10 but ADC.... Can someone test on an 8 Bit AVR board?

So I get correct readings with this line for LPC1786 / MKS SGEN L:
static inline float raw_to_mm(const uint16_t v) { return v * (3.3f / 16) * RECIPROCAL(float(MAX_RAW_THERMISTOR_VALUE)); }

How can we make this work for all boards?
3.3f can be replaced with HAL_ADC_VREF  (defined as 3.3 in Marlin/src/HAL/LPC1768/HAL.h).
Not sure about the divider, though. 
There is a "#define HAL_ADC_RESOLUTION     12" in HAL.h for LPC1768.
and "#define HAL_ADC_RESOLUTION 10" in HAL.h vor AVR. But I still do not understand why I have to divide by 16.

RicardoGA, 2021-07-08:
hi, i´m having the exact same issue with a BTT skr 2 i connect the filament width sensor and the readings are wrong

closed 2023-03-22 by thisiskeithb