MarlinFirmware/Marlin #19531: [FR] Force write marlin version and libraries to page

Tom-Neverwinter, 2020-09-27:
### Description

Force write, last version of marlin and libraries to page in code. no more random git forks that are for a build of marlin and lack any version information. 

reducing the need to literally reverse engineer poorly documented builds abandoned on git and download sites. 

Attempt to resolve issues with abandoned legacy devices, by increasing documentation.

### Feature Workflow

1. [First Action]
       user gets code. forks code. does work on code. 
2. [Second Action]
     code is abandoned. legacy hardware is forgotten.
3. [and so on...]
      user finds abandoned code. has same or similar hardware. however reading and understanding the code is hard without a reference to which version.

this forces information to be populated reducing the effort required to determine the build and other items.

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
N/A
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-09-27:
The main Marlin fork now updates [`Version.h`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Version.h) daily, so that will help a bit when looking at a fork.

bakaufman, 2020-10-11:
@thisiskeithb I am not getting Version.h updated. It is reading 2.0.5.4 from 2020-07-09 even though I have updated bugfix 2.0.x today using github desktop and pulled it into my printer version.

thisiskeithb, 2020-10-11:
> @thisiskeithb I am not getting Version.h updated. It is reading 2.0.5.4 from 2020-07-09 even though I have updated bugfix 2.0.x today using github desktop and pulled it into my printer version.

The update script only runs in this repo, so you'll need to add/update [`bump-date.yml`](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/workflows/bump-date.yml) to work in your fork.

bakaufman, 2020-10-11:
@thisiskeithb I don't know how to add that to my work. It is part of a workflow but I haven't used those before. Further, I am having the same problem with Marlin.ino no longer updating properly. A link or simple instructions would be greatly appreciated.

thisiskeithb, 2020-10-11:
> I am having the same problem with Marlin.ino no longer updating properly.

Sounds like something went fubar with your fork/branch when merging, but that's outside the scope of this feature request / issue queue. Please use the following resources for further troubleshooting:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

thisiskeithb, 2021-09-24:
Library version information is stored in [`platformio.ini`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/platformio.ini) and in [`/ini`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/ini).

Marlin bumps the distribution date for both [`Marlin/Version.h`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Version.h) and [`Marlin/src/inc/Version.h`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Version.h) now.

Both of which will be up to date the moment someone forks Marlin.

closed 2021-09-24 by thisiskeithb