MarlinFirmware/Marlin #19542: Printhead deviates from its GCODE path during printing then shows symptoms of hanging

motor24V, 2020-09-28:
_Might be an actual issue, hence posted here first. Kindly remove if the source is already known/suspected with great accuracy._

**TLDR:** as seen on the attached photo, the print moves get interrupted at a random point mid-print, then diagonal lines are extruded. Extrusion keeps happening, stopping the print from LCD results in scrambled moves, for example homing to the wrong corner (direction) at a wrong speed. Interestingly enough, the direction of the diagonal lines seems to be random as well. Like if the processor lost its knowledge of reading the GCODE language/firmware info. See details below.

**Configuration details:**
- Custom built large format printer
- BTT SKR v1.4 turbo
- Marlin-bugfix-2.0.x
- COREXY cinematics
- TMC5160 drivers on all axes
- Cura-generated Marlin-flavor GCODE read from the SKR's MicroSD card that also contains the config file

**Expected behavior**
- execute GCODE read from the MicroSD card and finish the printing process

**Observed behavior**
- at a random point during printing the printhead stops following the GCODE and the head goes haywire, always in a diagonal fashion comparable in length to the model being printed while continuously extruding molten filament (see attached image)
![20200928_234011835](https://user-images.githubusercontent.com/72047652/94489171-48c56780-01e4-11eb-9cc8-7ccbe974a41a.jpg
)

**Steps to reproduce**
- Could not identify, happens at every GCODE sliced with CURA, sometimes 10 minutes into printing and sometimes 1 hours, but never finishes

**Attempted resolution**
- increasing BLOCK_BUFFER_SIZE to 32 from the default

**Steps to work around**
- none found, so far as the source of the failure is unknown to me

[motor24v_config.zip](https://github.com/MarlinFirmware/Marlin/files/5295186/motor24v_config.zip)

thisiskeithb, 2020-09-28:
Try using a different SD card and/or slicer to rule out those as a possibility.

ellensp, 2020-09-29:
"always in a diagonal" this is what happens when one stepper motor stop moving on a corexy.  
Take note of which one and check its wiring and swap the stepper driver. (with a different axis, see if fault moves)

boelle, 2020-09-29:
> and check its wiring

^^^ What he said, that is what has happened to me a lot on my A motor (X connector) one of 4 wires where loose and both moved but one with more noise. Had to recrimp and use a new dupont housing/connector

motor24V, 2020-09-29:
Thanks for all the feedback. I am asking for your patience on this one, soon I'll have proper time to test everything you recommended. Hopefully I'll be able to solve this issue and indeed it won't be a software bug (now I suspect it is definitely not).

motor24V, 2020-09-29:
I am closing this issue as the problem seems to be gone. FYI I installed heatsinks on my XY motors so they run at around 50°C and this resulted in a 3 hour print finishing as intended. I think it was an overheat issue, afterall.

closed 2020-09-29 by motor24V