MarlinFirmware/Marlin #19543: Default Examples bad VERSION in Config.h and ADV.H files!!!

mikemerryguy, 2020-09-28:
Using Default Examples from Last Night
Config.h
Line 
41 #define CONFIGURATION_H_VERSION 020006
41#define CONFIGURATION_H_VERSION 020007
Config_adv.h
Line 
36 #define CONFIGURATION_ADV_H_VERSION 020006
36 #define CONFIGURATION_ADV_H_VERSION 020007

Sanity Check Fails unless updated

Using Ender 3
SKR mini E3 1.2 
STM32F103RC_btt_512K env in INI

Just a HEADS UP if anyone has issue compiling!!!

thisiskeithb, 2020-09-28:
Those configs were updated 15 hours ago and now have the correct version number.

https://github.com/MarlinFirmware/Configurations/blob/b5b5d00406e3e45810ddd41ff88e052b194d4ea8/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%201.2/Configuration.h#L38

https://github.com/MarlinFirmware/Configurations/blob/b5b5d00406e3e45810ddd41ff88e052b194d4ea8/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%201.2/Configuration_adv.h#L33

closed 2020-09-28 by thisiskeithb

mikemerryguy, 2020-09-29:
@thisiskeithb dang.... i downloaded maybe 18 hours ago!!!

Thanks!!!

This hobby IS FAST!!!!!!!!!!!!!!!!